pwri-opera/OperaSim-PhysX #3: [New Feature request] TFサポート

matsuren, 2023-02-27:
Navigation stackを使う際にTFが出ているとありがたいです．
具体的には[Robotics-Nav2-SLAM-Example](https://github.com/Unity-Technologies/Robotics-Nav2-SLAM-Example)の[ここ](https://github.com/Unity-Technologies/Robotics-Nav2-SLAM-Example/blob/main/Nav2SLAMExampleProject/Assets/Scripts/ROSTransformTreePublisher.cs)にTFのコードがあるようですが，
そのまま使えるか，ROS1でも同様に使えるかは私のほうでは検証できていません．

maopotcha, 2023-03-17:
リクエストありがとうございます。
対応させていただきたいと考えておりますが、ご要望の内容について、少々確認させてください。

TFトピックですが、具体的には、どのフレームからどのフレームへの座標変換が欲しいとお考えでしょうか。
(1) マップ上に固定された座標系→各建機のbase_link座標系
(2) 各建機内のリンク間の座標系
(3) 建機上のリンクに固定されたセンサ等の物体の座標系
上記の(1)のみかとは推察するのですが、(2)(3)或いはその他のものについても必要でしょうか。
(2)(3)について、OPERAにおいては（多くのROSを用いたロボットシステムと同様）urdfで記述された建機ロボットのモデルファイルからパラメータを読み出し、tfへpublishする[robot_state_publisher](http://wiki.ros.org/robot_state_publisher)を用いるのが通常であり、その入力となる/joint_statesトピックをシミュレータから出力しているため、基本的には不要と考えております。
(1)についても、各建機の/poseについては出力しているため、これを基にtfへのbroadcastを行うノードをros側に構築すれば、ひとまずシミュレータ側は変更なく対応可能かと思います。
（これまで、土木研究所ではこのような使い方をすることが多いです）

matsuren, 2023-03-26:
ありがとうございます．
私の意図としては
> (1) マップ上に固定された座標系→各建機のベース座標系

で具体的には`./odom`からロボットの一部，例えば`/footprint`や`/base_link`までのTFを意図していました．

個人的な実装の仕方として，
[turtlebot3_fake.cpp](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/master/turtlebot3_fake/src/turtlebot3_fake.cpp)のようにOdomを出すNodeにTFも同時に出させることが多かったのでコメントいたしました．そのほうがOdom msgとTFのFrame_idを合わせやすいのかなぁとも思っているのですが，正直TFは私は雰囲気で使っているのでそこまで詳しくないので参考まで．
ROS側でOdomをsubscribeしてTFをpublishする形だと，建機数を増やしたときにROS側も変更が必要になるデメリットもあるかなぁと．（そこまでいいだすと，理想はrobot_state_publisherもUnity側でやりたくなるのできりがないですが）

あと，TFが必要な状況として，ロボットに積まれたLRFの点群の座標変換を考えていました．ロボットLocalizationでは必須なことですが現状，`/pose`で絶対位置が取れてしまっているので私は緊急では使わない予定です．（リクエストしておいてすみませんが，そこまで優先度が高くないタスクでした．Issueを閉じても構いません）
今後，LRFやLIDARを載せてLocalizationも含めたNavigationを考慮する際はあったほうがよさそうです．（が，ROS側でも対応できるのは同意です．）

matsuren, 2023-03-26:
> ROS側でposeをsubscribeしてTFをpublishする形だと，建機数を増やしたときにROS側も変更が必要になるデメリットもあるかなぁと．（そこまでいいだすと，理想はrobot_state_publisherもUnity側でやりたくなるのできりがないですが）

逆に，ROS側からUnityのServiceを叩いて建機を`Spawn`させたほうがいい気もしてきました．
別issueに建てさせていただきます．

maopotcha, 2023-03-28:
ご回答ありがとうございます。
意図が(1)である(/odom->/base_link 又は/footprint)旨承知いたしました。
こちらでも改めて要否検討をしてみます。

基本的にはシミュレータ側からはセンサの模擬的な出力情報を出す形（未実装ですが）のみとし、その出力情報を元にした自己位置推定や、その結果を/tfに出力するのはROS側の処理に任せる思想ですので、このままCloseかも知れません。

いずれにせよ、改めて検討の上、結果をここにコメントするようにします。

yosuke, 2023-11-17:
@maopotcha @matsuren 
issueの元々の意図からはズレているかもしれませんが、重機の数が増えたときに柔軟に重機の現在位置を取得できる方法が欲しい、と思ったので、ground truth位置をTFとしてpublishするスクリプトを実装してみました。
https://github.com/pwri-opera/OperaSim-PhysX/commit/51e2b239a4705294b0afc593fcfb9f5709097349

odomに関してはROS側でwheel odometryをGPSと融合したり、SLAMを使って精度を上げたりして最終的にTFに出したいと思うので、シミュレータ側では出さない形にするので賛成です。

上記のground truthの実装もTFの上ではフレーム名にprefixなどつけて、推定値のTFの名前と被らないようにしておくと良いかもしれないと思っています。

yosuke, 2023-11-17:
ground truthについてはフレーム名にsuffixをつけて区別できるようにしておきました。
[image]

yosuke, 2023-11-17:
@maopotcha 
プルリクエスト作成しておきました。デフォルトでは無効にしてありますので、マージくださいませ。
https://github.com/pwri-opera/OperaSim-PhysX/pull/33

maopotcha, 2023-11-20:
@yosuke 
ありがとうございます。動作確認取れましたので、Mainブランチへマージします。

maopotcha, 2023-11-20:
@matsuren 
本件、TFのサポートに関するものについては、上記の内容にて意図通りでしょうか。
もし改善すべき点がありましたら、お知らせください。
今月（2023年11月）中に、何も返答なければ、本件は一旦クローズとさせていただきます。

maopotcha, 2023-12-07:
@yosuke 
本件、ROS SettingからprotocolをROS2にするとコンパイルエラーになります。
おそらくGroundTruthTFPublisher.cs中のtimeの型がROS1前提で記述されているからではないかと推察するのですが、ご確認いただけますでしょうか。修正いただければ幸いです。

--- 以下、Consoleに表示されるErrorメッセージ
Assets\Scripts\GroundTruthTFPublisher.cs(62,40): error CS0266: Cannot implicitly convert type 'uint' to 'int'. An explicit conversion exists (are you missing a cast?)

yosuke, 2023-12-08:
@maopotcha 
ROS2でのコンパイルの問題、修正完了しました（hotfixなので、メインブランチにマージもしておきました）。

maopotcha, 2024-02-08:
ご対応ありがとうございました。本件、クローズにします。

closed 2024-02-08 by maopotcha