pwri-opera/OperaSim-PhysX #7: [New Feature request] Spawnサポート

matsuren, 2023-03-26:
#3 に関連して`Spawn`サポートについて挙げさせていただきます．

現状は，以下のように建機の数を変更する際にUnity側とROS側の両方を変更する必要があります．

> ROS側でposeをsubscribeしてTFをpublishする形だと，建機数を増やしたときにROS側も変更が必要になるデメリットもあるかなぁと．（そこまでいいだすと，理想はrobot_state_publisherもUnity側でやりたくなるのできりがないですが）

Gazeboのように`Spawn`サービスがあると，Unity側を変更することなく，ROSのlaunchファイルだけで建機の数や位置を調整できると思いますが，いかがでしょうか？そうすると#3 の対応は不要になるかと思いました．

maopotcha, 2023-03-28:
ご意見ありがとうございます。
ご要望の確認ですが、ROSのメッセージ通信の枠組みを用いて、ROS側からメッセージを発行したら、Unity側のworld上に目的の建機(Unity側にPrefabとして予め登録してあるもの)が、目的の位置に`Spawn`されるというイメージでよいでしょうか。この程度の機能でよければ、おそらく実装可能だと思います。

もしよければ、`Spawn`時にUnity側に指定したい条件や要求仕様を箇条書きで列記いただくと、参考になりますためありがたいです。（内容によって、対応できない・難しいものがあることはご理解ください）

maopotcha, 2023-04-07:
@yosuke 
本件、仕様検討を含めてご対応をお願いできますでしょうか
ROS側からserviceをcallしたらUnity側にIDおよびnamespaceでrequestされた建機が、requestされた状態量（/pose・/joint_state)で`spawn`され、resultが返される、といった枠組みが良いかと考えていますが、その辺りも含めてご検討いただければ幸いです。

matsuren, 2023-04-07:
@maopotcha
連絡が遅れてすみません．私達の方でも検討しておりまして，GazeboのSpawnでできることが同様にできると嬉しいです．
また，強化学習にも使えないかなぁとも思っており，`reset`コマンドで建機の全破壊できて，Play中でも建機の再配置できるとありがたいです．（ゲームオーバーしたらやり直しみたいな機能は一般的になのでUnityの機能であると思うので`reset`でそれを呼ぶだけのイメージでいます）

また，建機を決められた位置に出現される際に，現状は違う建機で原点がずれているのを直す必要があるかと思います．現状は例えば`zx120`と`c30r`でpositionを(0, 0, 0)にしても同じ位置に現れません．

おそらく以下のスクリーンショットにあるようにbase_linkの位置がずれているのが原因だと思いますので，もし意図的なものでなければ直していただけると助かります．（上下方向のオフセットは接地させるためにひつようかもしれませんが，前後左右方向がずれているのが気になります．）
（前のバージョンのもので最新のバージョンでは直っている可能性がありますが以下がスクリーンショットです）
[image]

yosuke, 2023-04-10:
@matsuren @maopotcha 
gazeboのspawnは以下の定義ですが、model_xml部分にurdfかsdfが入るということになっており、ここをまともに対応するのは大変と思われる（内部でURDFImporterを呼び出す必要がある）ため、まずは簡易的な実装（xml内のnameアトリビュートからモデル名を判別してspawnする）にする予定です。
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/noetic-devel/gazebo_msgs/srv/SpawnModel.srv
このシミュレータの名前はgazeboではないのですが、ROSトピック名をgazeboと同じにしておくとspawn_modelコマンドがそのまま使えて良さそうですね。

yosuke, 2023-04-10:
@matsuren @maopotcha
zx120のbase_link位置がおかしい件も情報ありがとうございます。確かにこのモデルだけトップレベルのtransformとbase_link位置にズレがあります。
先日、同じ問題に気づいて修正した内部バージョンもあるのですが、公開バージョンも修正しておきます。

yosuke, 2023-04-10:
「モデルを出現させて関節角度を設定させる」に関しては、SpawnModelサービスが呼ばれてから、SetModelConfigurationサービスが呼び出される。この二つのサービスについては対応したい（他のサービスについては追って検討）。
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/noetic-devel/gazebo_ros/scripts/spawn_model#L195-L203

yosuke, 2023-11-17:
@matsuren @maopotcha 
本件、試しに実装してみましたが、案外問題なく動いています。
https://github.com/pwri-opera/OperaSim-PhysX/commit/b4950b02586fe15c8c01c2fcb1fa08b64bf1ac44
各種コントローラーやpublisherのROSトピック名をgameObject名に応じて設定するように変更する必要があるので、その部分が作業できたらプルリクエスト送ります。

yosuke, 2023-11-17:
spawnされたgameObjectの名前に応じて動的にROSトピック名を設定する変更も実装完了。もう少しテストしてみてから完成予定。

yosuke, 2023-12-01:
@matsuren @maopotcha 
以下のブランチでテスト可能な品質で動き始めています。もしよろければ試してみてご意見ください。
https://github.com/pwri-opera/OperaSim-PhysX/tree/feature/spawn_model

以下のようなコマンドで各ロボットがspawnできます。
```
rosrun gazebo_ros spawn_model -model test -urdf -file ic120.urdf -z 20
```

URDFファイルを指定していますが、Unity側ではあらかじめ作ってあるロボットのprefabからspawnしているので、URDFの内容は直接的には反映されない制限があります。この点についても許容可能かご意見いただけると幸いです。

maopotcha, 2024-02-08:
@yosuke 
ご対応ありがとうございます。私の手元ではROS1で動作確認取れました。
私は基本的にこの仕様で良いかと思います。

@matsuren 
いかがでしょうか。ご意見ありましたら今月中（2024年2月中）にいただけますでしょうか。
特になければ、この仕様でMainブランチに取り込もうと思います。

@yosuke 
今月の意見を待って、特に問題なければPull Requestをいただけますでしょうか。
現状のままだとMainブランチのSampleSceneとConflictが生じるので、解消もしていただけると助かります。
因みにROS2 HumbleでこのSpawn機能は使用可能な見込みでしょうか。
（Humble版のgazebo_rosではspawn_entity.pyを用いてROS1と同じようなSpawn操作ができるようですね）

yosuke, 2024-02-21:
@matsuren @maopotcha 
フィードバックありがとうございます。ROS2対応もそんなに大変ではないので、`SpawnEntity.srv`を使う形で実装しておきます。

UTFの`0x00a0`は、nbsp（HTMLの空白記号）のようですね、、、。
githubのweb画面から他で書いたコードをコピペしていたので、そこで混入したのかもしれません。

mainブランチからだいぶん差分が出てマージできなくなっているので、別ブランチを作って清書します。

matsuren, 2024-02-21:
@yosuke 
ありがとうございます．どうぞよろしくお願いいたします．

closed 2025-03-12 by maopotcha