pwri-opera/OperaSim-PhysX #10: 地面から建機への作用力算出機能を追加する

maopotcha, 2023-04-06:
@yosuke 
本issueについて対応をお願いいたします。

### 概要
建機が地面と接触した際に地面から建機が受ける力をシミュレートできるようにする

### 目的
建機の挙動およびアクチュエータの出力を現実に近づけるため

### やること
- 地面の粒子化範囲を拡大し、粒子化されたエリアの外側に対して力学的な拘束力を設定する
- (他にも良い案があれば検討をお願いいたします）

### その他
- AGX DynamicsやVoltex Stadio等、商用シミュレータの技術ドキュメントや実装方法を調査すると参考になりそう
- 商用シミュレータほどの精度や性能は不要

yosuke, 2023-04-11:
現状、粒子シミュレーションを実現するために、
1. 地面とバケットの間の反力の発生をオフにする（干渉のみ検出）
2. 干渉した部分に粒子を発生させる（粒子はバケットとの間に反力が発生する）
という実装をしているため、地面との反力が発生しない。

AGX Dynamics版については地面とのインタラクションを計算するネイティブ関数が呼び出されているようだが、そこそこリアルで簡易に実装できる手法を論文などで調査予定。
https://github.com/Algoryx/AGXUnity/blob/master/AGXUnity/Model/DeformableTerrain.cs#L241-L255

yosuke, 2023-04-11:
DEMを使った実装などもあるが、処理が重そう。

この論文の方法は、実用時間で動きそうな感じ。
https://arxiv.org/pdf/2107.04314.pdf

yosuke, 2023-04-28:
上記論文のKoopman-DFLを用いた方法はコードも公開してくれている。ライセンスはBSD
https://github.com/Darbeloff/DFL

yosuke, 2024-01-05:
地面に干渉した際の反力の方向をUnityのAPIで取り出そうとしていたが、バケットは地面にめり込まなければならないため、OnTriggerイベントで干渉を検出している。反力についてはOnCollisionイベントでないと取り出せない（計算されていない）ようだ。

OnTriggerの場合でも接触点は得られるので、その方向から反力方向を計算してしまえばよいか。もしくは、OnCollisionで反力をゼロに近く設定できればそれでもOK。

yosuke, 2024-01-05:
今まではバケットを地面をすり抜けるように設定して、OnTriggerで干渉を検出していたが、地形の高精度化が無事にできたので、干渉が発生次第、地面をへこませることができればOnCollisionを使う形に戻しても問題なさそう。要検討。

yosuke, 2024-01-05:
TriggerからContactに変えると接触点とともに反力ベクトルがきれいに出力されていることは確認。
[image]

maopotcha, 2024-02-08:
ご対応ありがとうございます。本件、一旦クローズにします。

closed 2024-02-08 by maopotcha