pwri-opera/OperaSim-PhysX #8: ROS2対応

maopotcha, 2023-04-06:
@yosuke 
ROS2に対応可能なI/Fを選定していただけますでしょうか。
(1) ROS-TCP-Connector/Endpoint
(2) ros2-for-unity
(3) dds for unity?
etc
上記の中で、現状のsim_physxとして相応しい物はどれでしょうか。
現状は(1)を選定しており、ROS/ROS2供に通信可能となっています。
アウトプットイメージとしては、まずは上記の候補を縦軸、評価項目を横軸に取って星取り表のような形で
まとめていただければと思います。
横軸にどのような項目をセレクトすべきかも含めて、検討をお願いします。
（まだ実装上の変更や、実験を伴う比較は不要です）

yosuke, 2023-04-21:
まずは、各実装の詳細と特徴をリストします。
最後に星取表の形でまとめます。

yosuke, 2023-04-21:
(1) ROS-TCP-Connector/Endpointの場合
--------------------------------------------
Unity公式のROS-TCP-ConnectorはROS2に対応している。

Connector/Endpoint間はTCP通信しており、Unity側から見るとROS1/ROS2でプロトコルは変わらない。
ROS2対応のためにEndpoint側に以下の変更が加えられている。
https://github.com/Unity-Technologies/ROS-TCP-Endpoint/commit/127a8a0301a3f04bac4476a55ec45d2836347ad3

メリット：

- Unity公式なので、長い間サポートされる安心感がある。
- Unity側からはROS1/ROS2を意識することなく使えるので、コードが共有することで開発効率を高められる

デメリット（注意点）：

- ROS2利用のメリット（リアルタイム性やQoS機能）はEndpointのコード上で有効化されていない。あくまでプロトコルとしてROS2を使うことができるだけ。例えば、LiDARデータをQoS制御つきでシミュレートする、などができない（Endpointにパッチを当てればできるかもしれない）。
- UnityとEndpoint間はTCP通信なので、この部分はQoS制御などはできない。

yosuke, 2023-04-28:
(2) ros2-for-unityの場合
-----------------------
Robotec.ai社が開発しているROS2対応ライブラリ。

サードパーティー製だが、GitHubスター数が252あるなど人気が高い。
https://github.com/RobotecAI/ros2-for-unity

内部ではros2cs(ros2_dotnetのフォーク実装)を使っており、C++ネイティブ実装のros2 client library(rcl)を内部的に呼び出している。
https://github.com/RobotecAI/ros2cs

メリット:
- 内部的にros2 client library(rcl)を使っているので、QoS含むROS2のフル機能が利用できる
- Unityから直接ROS2プロトコルが使えるため、ROS-TCP-Endpointのようなブリッジを必要としない

デメリット（注意点）:
- サードパーティライブラリであるため、長期サポートにやや不安が残る。ただし、現在は公式となっているROS-TCP-Connectorも、元はシーメンス社が開発していたROSSharpを公式化したものであり、本ライブラリものちに公式化される可能性は十分にある
- C++ネイティブ実装のros2 client library(rcl)を内部的に呼び出しているため、クロスプラットフォーム動作しない。現在、LinuxとWindowsでのみ動作し、Macはサポートされていない

yosuke, 2023-04-28:
(3) dds-for-unityの場合
-------------------------
https://bitbucket.org/edhage/dds-for-unity/src/master/

デメリット:
- 最終更新日が2021年10月となっており、開発が止まっている。
- 内部ではVortex OpenSplice DDSを使っているが、OpenSpliceはROS2の公式サポートから外されている。 https://discourse.ros.org/t/opensplice-move-to-eclipse-cyclone-dds-in-ros-2/12370/3

maopotcha, 2024-02-08:
本件、Unityの公式に合わせて(1) ROS-TCP-Connector/Endpointの仕組みを用いることとしたいと思います。
Unity側で動きがあったら、そちらに追従する方針として、本件は一旦クローズにします。

closed 2024-02-08 by maopotcha