pwri-opera/OperaSim-PhysX #9: 地形の変形の精緻化

maopotcha, 2023-04-06:
@yosuke 
ショベルで地面掘削をすると、現実にはあり得ない地形の隆起が生じることがあります。
(1) これを防止するような地形シミュレートモデルの修正
(2) 地形シミュレートのenable/disableをGUIで簡単に切替えられる機能
を追加していただけないでしょうか。

yosuke, 2023-05-08:
Mudシミュレーションの際にも参考にさせてもらったDaniel Holzさんが以下の論文を書いている。
重いDEMと高速なGridベースのシミュレーションを組み合わせるアプローチ。
https://www.researchgate.net/publication/221622677_Soil_Deformation_Models_for_Real-Time_Simulation_A_Hybrid_Approach

yosuke, 2023-05-08:
参考になるかもしれない実装。
https://github.com/edualvarado/unity-footprints

yosuke, 2023-06-29:
Unityの標準機能でThermal（堆積物）の隆起シミュレーションがあるらしい。
https://docs.unity3d.com/ja/Packages/com.unity.terrain-tools@5.0/manual/erosion-thermal.html
砂、各種粘土質、Dry Ashなど堆積物の特性を設定して安息角を用いて繰り返し計算する実装。
プログラム的に呼び出す方法がないか調べ中。

yosuke, 2023-06-29:
以下の論文も参考になりそう。安息角を使った非常に簡単なアルゴリズムでリアルなシミュレーションができている。
https://arxiv.org/pdf/1908.02745.pdf
ブルドーザーなどによる地形の変形にも適用可能とのこと。

yosuke, 2023-06-30:
Unity標準機能の実装が見えるようになっていた。
https://github.com/needle-mirror/com.unity.terrain-tools/blob/master/Editor/TerrainTools/Erosion/ThermalEroder.cs
材質によるパラメータ設定は以下の通り。
https://github.com/needle-mirror/com.unity.terrain-tools/blob/master/Editor/TerrainTools/ThermalErosionToolOvl.cs
繰り返し計算部分がシェーダーで実装されているなど非常に良くできている。
https://github.com/needle-mirror/com.unity.terrain-tools/blob/master/Editor/TerrainTools/Compute/Thermal.compute
ほとんどのクラスがinternal設定されているが以下の方法で呼び出し可能とのこと。
https://qiita.com/mob-sakai/items/f3bbc0c45abc31ea7ac0

yosuke, 2023-06-30:
internal設定されたクラスへのアクセスは可能だが結構面倒くさい。
シェーダー部分にThermal Erosionのほとんどの処理が書かれており、シェーダーは以下のコードで外部からロード可能なので、その使い方をするのがよさそう。
https://github.com/needle-mirror/com.unity.terrain-tools/blob/master/Editor/TerrainTools/Compute/ComputeUtility.cs

yosuke, 2023-12-15:
シェーダーを使うと安息角を使った処理が非常に高速できる、が、その結果をTerrainDataに格納して`SyncHeightmap()`を実行するとかなり遅くなってしまう。
SyncHeightmap以外の方法でColllisionとVisualを更新するか、部分的に必要箇所だけSyncするなどの対処が必要と思われる。

yosuke, 2023-12-22:
SyncHeightmapを必要なエリアだけに適用するように最適化したところ、実用的な速さにすることができた。地形も自然に変形しているように見えるが以下のブランチでマージ前の最終チェック中。
https://github.com/pwri-opera/OperaSim-PhysX/tree/feature/terrain_restangle

MaruiYuta, 2024-02-13:
2/9~2/16 粒子半径を変動させシミュレータ挙動を確認する。
掘削作業時の粒子半径(Particle visual Radius)を初期値0.15から0.01毎に小さくしていき確認しました。"プログラム内で油圧ショベルの目標角度と実際の角度との差(角度偏差)をどこまで許容するか"と"粒子半径がどこまで小さくできるか(小さくできると粒子らしい挙動が見れてよい感じ)"です。油圧ショベルの挙動のみ焦点を当てた場合は角度偏差は0.1でも問題ありませんでした。実際に掘削動作を行うと0.3にしても動きが止まってしまうことがありました。粒子についてはParticle visual Radiusが0.05から粒子量が多くなりシミュレータにカクつきが見られました。粘性とリアルタイム性を考慮してDaniel Holzさんの論文を参考に実装していますが粘性の部分がまだ見て取れなかったです。(どちらかというと米粒がパラパラしているイメージ)。リアルタイム性に関しては各時刻において挙動が見られたため満たしていると思います。逆に値をものすごく大きくし実行もしました(0.9)。挙動は現実離れしたものでしたが処理に関しては重くなったです。よろしくお願いします。
[image]
[image]

yosuke, 2024-02-15:
@MaruiYuta 
Visual Radiusだけでなく、RockPrefabのSphere ColliderのRadius値も同時に小さくしてみてください。Visual Radiusは見た目の大きさだけが変わって干渉形状の大きさは変わりません。

MaruiYuta, 2024-04-27:
@maopotcha @yosuke @TaroABEJido
先日はありがとうございました。
現在調査段階ではありますが２つのタスクに取り組んでいます。実装の仕方が分からず思考中であるため良ければ参考になる資料、もしくは勉学すべき内容を教えていただければ幸いです。
以下、２点をまずは行えるようにしようと考えております。
(1)以前の実験のRosbagを使用して、Unity内の油圧ショベルを実機と同様の動きをさせたい。
(2)実験で取得したRosbagのgnss_compassトピックのメッセージを使用して現実の油圧ショベルの位置とUnity内の油圧ショベル位置を一致させたい。
(1)については以前までシミュレータで動作させていた時はcmdに指令値を流して実行していました。今回はIMU(g2_imuトピック)のorientation,angular_velocity,linear_acceletationを使用して同様の動作をunity内で実行したいと思っています。(Bucket部に搭載しているアブソリュートエンコーダが不調のため)
どうやってこのIMUを使用してunity内の油圧ショベルを動作させるかで悩んでいます。
(2)についてはgnss_compassトピックを使用してunity内の油圧ショベルの位置合わせを行いたいと考えています。
現実とシミュレータ内で考えると
関東９系→？？？(シミュレータ内のどこか)
関東９系から見た油圧ショベルの位置→gnss_poseトピック(x,y座標ベース)
油圧ショベルの姿勢(向き)→headingトピック
自分の認識は上記のような認識です。そのためまず関東９系とunity内のworld座標系の位置合わせを行いたいのですがどこを参考にしたらよいかで悩んでいます。

長文失礼しました。
ご意見いただければ幸いです。宜しくお願い致します。

TaroABEJido, 2024-05-01:
@maopotcha @yosuke @TaroABEJido @MaruiYuta 

連絡ありがとうございます．取り急ぎ質問への回答です．
> 今回はIMU(g2_imuトピック)のorientation,angular_velocity,linear_acceletationを使用して同様の動作をunity内で実行したいと思っています。(Bucket部に搭載しているアブソリュートエンコーダが不調のため) どうやってこのIMUを使用してunity内の油圧ショベルを動作させるかで悩んでいます。

IMU の情報から計算された joint state は `zx120/imubased/joint_states`   に出力されているのでそれを使用してください．
関節情報になっているので，あとは任意のノードを作成し，メッセージを分解して cmd [Float64] として出力すればよいと思います． 

また，コメントを読んだうえで気になった点や話を整理したい点について，いくつか確認したいことがあったので，列挙します．以下長文失礼します．

### 実機とシミュレータの合わせ込みについて
質問(1) にて Unity ないの油圧ショベルにて実機と同様の動きを実現したいとありますが，現状ではシミュレータ内ショベルは現実のショベルの動きを反映していません．念のために確認です．ですので Rosbag 内にあるショベルに対する目標値，例えば `zx120/swing/cmd` をシミュレータに与えても同様の動きは実現されないです．（実機におけるむだ時間等が再現されていないので）

今回は 実測された IMU より計算された各関節角度 を cmd に渡したいとのことでしたが，あくまでも物理シミュレータ内ショベルは内部に仮に設置しているPIDゲインで制御されている都合上，実機のそれとは状態が異なるため実機の動きは再現されないと思います．

物理シミュレータ内ショベルが実機に合わせられていない以上，現時点で質問(1) のようなことをしても結果の比較として使えるデータは得られないと思う反面，以下を試したい場合であれば有効であるかと思います．

* **ICP によるパラメータ調整の有効性の検証**
地面を再現する各パラメータの同定？のために 各パラメータを何らかの要素として ICP により 最適なパラメータを探ることをm圓井君としてはやりたいことであると私は理解しているのですが，その有効性の検証のための仮の実験として実機の動きをシミュレーションに反映させたいのであれば，現段階で 質問1 のタスクをやる意義はあるように感じます． 

### 実機とシミュレータのショベルの位置合わせについて
質問(2) にて `/gnss_compass` トピックを用いて ショベルの位置合わせを行いたいとのことですが，1) 地面を平面と仮定 2) スキャンした地形を Unity に読み込ませる かで gnss のデータを位置合わせに利用する必要性が変わるかと思います．

1.  **地面を平面と仮定**
物理シミュレーションにて 地面が平面の場合，どこを掘削しても結果は変わらず  gnss のデータは重要ではないと思います．むしろ実験の際に配置したコーンとショベル，および掘削穴の位置関係が重要で，物理シミュレータにおいてもそれぞれの相対位置が把握されれば良いと思います．

2.  **スキャンした地形を Unity に読み込ませる**
この場合は GNSS の測位結果による位置合わせをする必要があると思います．現在実装中なのですが シミュレータ内ショベル では 緯度経度も取得できるようになる予定のため，関東9系の x,y 情報は 特に不要です．GNSS は オープンソースの [image] で実現することを考えています．

TaroABEJido, 2024-05-01:
@yosuke @maopotcha @MaruiYuta 
とりあえず実験的に評価するのであれば 松坂さんが提案下さった，上記コメントhttps://github.com/pwri-opera/OperaSim-PhysX/issues/9#issuecomment-2087929141 のやり方でも良いと私も感じました．

MaruiYuta, 2024-05-07:
1. @yosuke @maopotcha  @TaroABEJido 
回答していただきありがとうございます。
現在はimubase/joint_state内のswing_joint,boom_joint,arm_joint,bucket_jointをそれぞれ別のトピックに流すノードを作成し動作する部分まで行えている状況です。より忠実な再現をシミュレータ下で実行するためご教授いただいた関節ゲインを大きくする,重機の各リンクを軽くする設定について質問したくcommentを書いています。
以下に疑問点を記載します。回答いただければ幸いです

1. 関節ゲインを大きくする。
→これはREADME.mdに記載されている関節可動値のパラメータまたは関節制御が振動的になった際のシミュレーションパラメータのチューニングに記載されている内容のことではなくPIDゲインのことという認識で正しいでしょうか。もし後者だった場合、その制御プログラムはどこに記載されているのか教えていただきたいです。

2.重機の各リンクを軽くする
→どこにその設定値があるのか教えていただきたいです。

宜しくお願い致します

[image]

yosuke, 2024-05-07:
@MaruiYuta 
左のツリー構造を展開してzx120の各リンクをクリックしてください。
右のパネルに「Mass」や「xDrive」という欄が出てきます。Massを少なく、xDriveの各パラメータをかなり大きく設定してみてください。
[image]

maopotcha, 2024-05-08:
@MaruiYuta 
今回は一旦、進めていただいているようにシミュレータのショベルモデルの挙動を実機に無理やり合わせる（強制変位を与える）方向で良いかと思いますが、本来はショベルモデルの挙動が実機に合うように、上記のパラメータを同定するステップが必要だと考えます。それでも実機と合わせ込み切れない場合はモデルそのものの修正を行ってパラメータ同定を行う、というサイクルを何回か繰り返してより良いモデルとモデルパラメータの組み合わせを探る、というステップも研究計画の中に組み込むことを検討いただければと思います。このようにして油圧ショベルの挙動を実機とシミュレータモデル間で限界まで合わせ込んだ後、地形変形のモデル改良とパラメータ同定のサイクルを繰り返す、という流れを今後の進め方として検討・想定いただけますでしょうか。

MaruiYuta, 2024-05-08:
@maopotcha @yosuke @TaroABEJido 
分かりました。では挙動の合わせこみから進めていきます。
> 本来はショベルモデルの挙動が実機に合うように、上記のパラメータを同定するステップが必要だと考えます。
はい自分もそのように感じました。
そのステップを行いたく昨日yosuke様からご教授いただいたパラメータ値について思考しておりました。
私なりの仮説を立てました。以下に記載します。

1. massを小さくする
→質量を小さくすることで与えられるトルクを反映しやすくする役割がある。そのため与えられた指令値(target)に対して近づきやすくすることができる

2. Stiffness,Dampingを大きくする
→これはトルク導出に使用される係数であり大きくするとその分トルクが大きくなるため、cmd指令値に対して近づきやすくすることができる(この点がmaopocha様がおっしゃっていた"強制変位を与える"と考えた)

間違っていた場合、ご教授若しくは参考となる資料等をご教授いただければ幸いです。
また合わせこみに対する評価手法についても思考段階であり何か良い手法があればご教授いただきたいです。
今思いつく限りではROSBAGから出力された各jointの目標値と実際のjointの数値誤差を時々刻々計算しそれを別変数で累積誤差として保持しておく、保持された累積誤差が小さければ小さいほど誤差が少なく忠実に再現できているというものです。

宜しくお願い致します。

maopotcha, 2024-05-09:
@MaruiYuta 
> 本来はショベルモデルの挙動が実機に合うように、上記のパラメータを同定するステップが必要だと考えます。

ここで言った「ショベルの挙動を実機に合わせる」、ということと強制変位を与える（＝@yosuke の”ミラーリングする”）は全く別です。massを単純に小さくする、stiffness, dampingを単純に大きくする、は共に後者のためのアプローチです。今回はrosbagのjoint_states又はstateの値をシミュレータ内のショベルへのcmdにリマップして入力することになると思います（rosbagのcmdをシミュレータのcmdに入れるのではありません。念のため）

前者については[こちら](https://tsukuba.repo.nii.ac.jp/record/47037/files/DA08527.pdf)を参照してみてください。そもそも現在、シミュレータが有しているショベルのモデルと文献内のモデルは全く異なると思います。実機との合わせ込みにおける評価指標としては、実機の関節角度の時間プロファイルと、シミュレータの関節角度の時間プロファイルとの誤差の実効値でよいかと思います。但し、この評価についてはまずは単純な１軸ごとのステップ入力等の条件での実測データに基づいて行うべきだと考えます。

MaruiYuta, 2024-05-15:
@maopotcha @yosuke 
joint_stateの値をcmdにリマップし関節角を制御しているのですが、target(目標指令角)とrotation(駆動角度)が合わず悩んでおります。stiffnessの値を大きくしトルクの大きさを上げ強制的に変位するようしたつもりなのですが...。rotation.xはどのcsプログラムで度数変換を行っているのでしょうか？
現状。terrainとバケットの接触による粒子生成により関節角度がtargetに時刻通りにならない問題に関しては粒子半径を大きく(以前自身が行った粒子径におけるシミュレータ挙動実験を参考)することで解決していていると思っております。
[image]

yosuke, 2024-05-16:
@MaruiYuta 
rotationのxは「ワールド座標上での姿勢」なので、多関節ロボットの関節の現在角としては直接的には使うことができません（計算すれば使えますが結構複雑な計算が必要です）。

現在の関節角が欲しい場合、「ArticulationBody」の「jointPosition」プロパティから簡単に取り出せます。
下のスクリプト上で実装されているのですが、すでにROSのjoint_statesトピックとしてpublishもされているので、そちらを使ってみてください。
https://github.com/pwri-opera/OperaSim-PhysX/blob/main/Assets/Scripts/JointStatePublisher.cs#L66-L68

MaruiYuta, 2024-05-20:
@yosuke @maopotcha @TaroABEJido 
フィードバック有難うございます。
以下のプログラムでシミュレータ内の関節角を取得しその二乗誤差を計算しています。
https://github.com/MaruiYuta/opera_confirm/blob/feature/BehaviorMatch/zx120_sample/src/internship_task/src/rosbag_cmd_Publisher.cpp
また映像とrosターミナルの様子は以下になっています。(issueでのアップができなかったため私のリポジトリから失礼いたします)
https://github.com/MaruiYuta/opera_confirm/tree/feature/BehaviorMatch/movie
現状、rosbag通りの挙動を再現できていません。申し訳ございません。
強制変位を与えても実物と同様の動きをしない(特にBucket)原因がわからないです。プログラムを動かして二乗誤差を見るとbuket値だけが大きく変わっているため同様の動作が実行できていないことはわかるのですがこれはstiffness,dampingの関係なのでしょうか。長文失礼しました。
ご教授いただければ幸いです。宜しくお願い致します

yosuke, 2024-05-22:
@yosuke 
映像、見ました。
バケットの角度（だけでなくアームの角度も）にオフセットがかかってしまっているように見えます。
バケットの関節可動角限界に達しているため、それ以上、角度を変えることができないため誤差値が発生して蓄積していそうです。
プログラムを見た感じですが、rosbagからの関節角度がreturn_cmdに格納されているものの、cmdとしてpublishしているのはjoint_cmdになっているように見えますが、これはあっているでしょうか？

MaruiYuta, 2024-05-22:
@yosuke 
フィードバック有難うございます。
ArticulationBodyのLower limit,upper limitをtargetの数値より大きく範囲をとっているため問題ないと思っています。Lower limit,upper limitで関節可動角限界を決められるという認識です。
プログラムに間違えがあり、jointstateDiffCallback関数のjointstate型のnameがbucket_jointならjoint_return_pos[BUCKET]に格納する部分をnameがboom_jointならになっていました。そのためboomの関節角が格納されてしまい、角度誤差が大きくなってしまっていました。
https://github.com/MaruiYuta/opera_confirm/commit/04b64a675c0026b8396e9ccc7ca7a89ec2ea7d50#diff-03ab7f843c2a1e8eb45bd8a254aaf99fabbb926bd60a684f0e8c21ea8209d5abL53
しかし角度誤差が小さくなった今現在もバケットが土をすくう動作が見られません。誤差は小さいので実機と同様に動くと思うのですが...
またプログラムについてはrosbagからの関節角度はjoint_cmdに格納し、その後unity内のシミュレータの関節角度をreturn_cmdに格納しrosbagから送信した値(joint_cmd)-シミュレータ内の現在の関節角度(return_cmd)で時々刻々の関節誤差を算出しているという流れになっております。問題があれば指摘していただければ幸いです。
また質問なのですがTransformのRotationとtargetは関係がないものでしょうか。私はRotationがtarget値に向かうようになっているという認識です。
実際の様子をupします。ご確認いただければ幸いです。
https://github.com/MaruiYuta/opera_confirm/tree/feature/BehaviorMatch/movie
ご多忙のところ申し訳ございませんがご教授いただければ幸いです。宜しくお願い致します。

yosuke, 2024-05-22:
@MaruiYuta 
各limitを大きくしても自己干渉で可動角は狭くなってしまいます。また、バケット関節に制御がかかっていそう（diffも小さい）なのにあさっての方向を向いているのがおかしいです。
rosbagに入っている角度はIMUから取得した角度、だと思うのですが、これが「親リンクとの相対角」ではなく「ワールド上での絶対角度」になっていたりすると、バケットで変なオフセットがかかっていることも説明できそうに思うのですが、、、。

@maopotcha @TaroABEJido 
このあたり、フォローしてもらえますか？

TaroABE, 2024-05-24:
上記件、返信が遅れてしまいすみません
対応致します

TaroABEJido, 2024-05-26:
@MaruiYuta 
とりあえず bagfile 内トピックの /zx120/joint_states  と /zx120/imubased/joint_state を確認してみたのですが，
 値の顕著な差は見られませんでした（bucket 関節角の情報だけは アブソリュートエンコーダの情報が正しくないため，それ以外ですが）
 もう少し私も確認してみます

TaroABEJido, 2024-05-27:
@maopotcha @MaruiYuta @yosuke 
上記物理シミュレーション上ショベルの件、そもそも、実機とシミュレーションで関節角度値が0となる姿勢が異なる可能性はないでしょうか？

気になったためコメントします

TaroABEJido, 2024-06-03:
@MaruiYuta 
その後，ショベルの挙動再現の件はどうなりましたでしょうか？
先週にかけてあまりフォローする時間が取れずすみません．

TaroABEJido, 2024-06-03:
@MaruiYuta @maopotcha 
試しに私の方でも動かしてみました．IMUの値を用いてもやはり bucket の動作に問題があるようです．
[image]

TaroABEJido, 2024-06-07:
@MaruiYuta @maopotcha 
画像二枚目について，建機の点群情報は必要ではないかもしれません．
詳細は別途打ち合わせしましょう

MaruiYuta, 2024-07-06:
@yosuke 
お世話になっております。調整しているパラメータについて質問です。
このパラメータ(粒子径,粒子間引力,粒子間引力距離)は何か先行研究を参考にし使用しているのでしょうか。また現実でこのパラメータは計測できるのでしょうか。
ご多忙のところ申し訳ありませんが回答していただければ幸いです。宜しくお願い致します

maopotcha, 2024-07-08:
基本的には以下の文献を参考にしたオリジナルのモデルとなっています。
また、現実でこのパラメータの計測は不可能です。

https://diglib.eg.org/bitstream/handle/10.2312/PE.vriphys.vriphys09.021-030/021-030.pdf?sequence=1&isAllowed=n

粒子径は見た目の細やかさと計算負荷を天秤にかけてユーザ側が決めるパラメータです。
粒子間引力も現実ではほぼ作用していない（万有引力くらい）です。現実にある土砂の粘性の特性をこれらのパラメータ設定で、模擬的に（誤魔化して）表現しているに過ぎません。

@yosuke
補足がありましたら、お願いいたします。

yosuke, 2024-07-08:
@MaruiYuta @maopotcha 
上記で認識あっています。

粒子間力ですが、粒子がパラパラに飛び散ってしまうのを防ぐため、万有引力よりはかなり強めにかかっています。

元論文でもシミュレーション結果と実際の掘削の写真を比較して「実際に近い」という定性的な評価しか行われていない（ただし、こちらの結果を見ても、粒子間力の追加で実際にかなり近くなるのは事実）ですし、パラメータの設定も手探りの状態なのですが、それを定量的に評価できるようになると画期的、というのが今回の話かなと思っています。

MaruiYuta, 2024-07-15:
@yosuke @maopotcha @TaroABE 
フィードバック有難うございます。
進捗として,cloudcompareで土研点群とシミュレーター点群をコーンでICP位置合わせし地形全体の点群平均距離(mean distance)粒径0.01～0.2(1cm～20cm)の時で算出しました。※particlevisualradius(visual,prefab)は同値に設定。資料に結果を載せております。ご確認頂ければ幸いです。
[Uploading 20240715.pptx…]()
現状,悩んでいる部分について質問させていただきます。
1.cloudcompareでコーンのみセグメントしICPを行うとコーンのみのRMSは算出できるが地形全体のRMSが算出できない。よってcloud distance toolを使用してmean distanceを算出している。(点群全体でICPするとコーン位置がずれてしまうため)
RMSで評価した方が良いか、mean distanceは評価指標として正しいか、誤りであればcloudcompareでコーン位置をICPした状態で地形全体のRMSを算出する手法をご教授頂きたいです。
2.stickForceのパラメータを調査しているのですが,シミュレータsceneを見るとはじくような挙動が見られました。(後ほど動画アップします)。引力を大きくするとはじく動作ではなく粒子同士が粘り気を帯びた様子が見られると思いました。この認識は正しいでしょうか。

上記の点、ご教授頂ければ幸いです。
今後,stickForceとstickForceDistanceのパラメータについても同様の手法で調査を予定しております。

宜しくお願い致します。

MaruiYuta, 2024-07-15:
@yosuke @maopotcha @TaroABE 
2で質問させていただいた実際のシミュレーション動画になります。
[OperaSim-PhysX 位置確認_stickForce - SampleScene - Windows, Mac, Linux - Unity 2022.3.4f1_ _DX11_ 2024-07-15 11-26-31.zip](https://github.com/user-attachments/files/16229592/OperaSim-PhysX._stickForce.-.SampleScene.-.Windows.Mac.Linux.-.Unity.2022.3.4f1_._DX11_.2024-07-15.11-26-31.zip)

TaroABEJido, 2024-07-16:
@MaruiYuta @maopotcha @yosuke 
> stickForceのパラメータを調査しているのですが,シミュレータsceneを見るとはじくような挙動が見られました

これは stick force をいじらない状態で発生したものでしょうか？ちなみに branch，feature/realistic_soil_particle_behavior 土の挙動ではこのような状態にはならないです．
（文脈から察すると，stickForce を小さな値にするとはじけるような挙動が発生したということ？）

TaroABEJido, 2024-07-16:
@MaruiYuta @maopotcha @yosuke
> cloudcompareでコーンのみセグメントしICPを行うとコーンのみのRMSは算出できるが地形全体のRMSが算出できない。よってcloud distance toolを使用してmean distanceを算出している。(点群全体でICPするとコーン位置がずれてしまうため)

cloud distance tool （CloudCompare の機能のこと？）を使用すると，コーンの位置や姿勢をフィットさせることができるということでしょうか？
mean distance がどのような値か分からないので，現状なんとも言えませんがどのような値か分かりますでしょうか？

MaruiYuta, 2024-07-16:
@TaroABE @yosuke @maopotcha 
particlestickdistanceを仮に1としたとき,はじくような動作が見られました。
cloudcompare内tool ICPで位置合わせしました。その後distance toolで点群平均距離(すべての点群に対して最近傍)を算出しています。

TaroABEJido, 2024-07-16:
@MaruiYuta @yosuke 
> particlestickdistanceを仮に1としたとき,はじくような動作が見られました。

ソースコード (BucketRock.cs) を見るに，particleStickDistance の距離を1.0 にすると，引力が働く範囲が 粒子直径を超えるので，隣合う粒子以外にも粒子間力が働くことになりそうな気がしました．
それがはじけるような挙動を起因させているかはよく分からないです．

ただ，粒子同士が particleStickDistance の範囲内で 遠くなればなるほど，粒子間力が大きくなるので，複数の粒子塊間に加わる力は増えそうな気がします（それが原因かはわかりませんが ...）．

TaroABEJido, 2024-07-16:
@MaruiYuta

> cloudcompare内tool ICPで位置合わせしました。その後distance toolで点群平均距離(すべての点群に対して最近傍)を算出しています。

了解です．どのような手段を使うにせよ，平均距離を測る前にコーン同士をマッチングできているかが重要と思います．コーンはマッチング出来ていますでしょうか？

MaruiYuta, 2024-07-17:
@TaroABE @yosuke @maopotcha 
[image]
コーン周辺部のみでICPで位置合わせしました。コーン上部を除いたのは位置合わせが定性的にできていなかったためです。
[image]
その後mean distanceを算出しています。

MaruiYuta, 2024-07-17:
@yosuke @maopotcha 
使用している引力の定量的な評価はされているのでしょうか。
宜しくお願いします

TaroABEJido, 2024-07-17:
@MaruiYuta 
位置合わせ後の結果ですが，掘削されていない地面を見ると，コーンから離れた位置になればなるほど，差が開いているように見えます．例えばですが，掘削後の点群における掘削びよって変化した部分を削除し位置合わせ を行うことで，この差が軽減されると思うのですがどうでしょう？
[image]

@MaruiYuta @yosuke 
> 使用している引力の定量的な評価はされているのでしょうか。
圓井君のいう定量的な評価がどのようなものかがイメージできませんが，
どのようなものでしょうか？

MaruiYuta, 2024-07-17:
@yosuke @maopotcha @TaroABE 
数値的に強弱が分かるとユーザー側も使用しやすいと思いました。そのため数値的にどのあたりが強く,どのあたりで弱いかという意味で評価できているかをお聞きしたかったです。

位置合わせについてもう少し詳細に教えていただいても良いでしょうか

yosuke, 2024-07-19:
@MaruiYuta 
出遅れてすいません。粒子が弾かれるような症状ですが、私の方でも以前再現し、`feature/realistic_soil_particle_behavior`で対策が実装してあります。

粒子間力と距離範囲ですが、どのあたりが良い値だ、という定量的な評価はできておらず、、、（粒子径によっても最適値が違うので）。
ただ、一般的な土の弾性・粘性・塑性については、既によく知られた値があるので、これを機会に、それらの値に近づけるには各パラメータをどう設定するのが良いか、シミュレータベースで調べてみたいとも思っています。
今回のような掘削動作ではなく、3軸圧縮試験のようなシミュレーションをすれば良いと思うのですが、私の方で検討してみます。

MaruiYuta, 2024-07-19:
@yosuke 
回答いただき有難うございます。後ほど自分の環境でも確認してみます。
シミュレーター上の粒子間力について数式に基づいて定義されているのでしょうか。もしくは一方の粒子モデルに対してもう一方の粒子モデルの方向に加わる力として実装されており数式的定義はされていないのでしょうか。よろしくお願いいたします

yosuke, 2024-07-19:
@MaruiYuta 
ここで近くの粒子から力を加えるベクトルを計算し
https://github.com/pwri-opera/OperaSim-PhysX/blob/main/Assets/Scripts/BucketRocks.cs#L159-L170
ここで実際の力を加えています
https://github.com/pwri-opera/OperaSim-PhysX/blob/main/Assets/Scripts/BucketRocks.cs#L200
上記はプログラム的な実装ですが、数式的な表現をすることも可能です。

MaruiYuta, 2024-07-19:
@yosuke
有難うございます。
ベクトル計算部分を確認していて理解できていない箇所があるためご教授頂ければ幸いです。
        public void Execute(int index)
        {
　　　//粒子モデルの位置
            var rock1 = positions[index];
　　　//3次元ベクトルの初期化
            var repulvector = new float3(0, 0, 0);
　　　//粒子数だけループ(positions.Lengthの定義が不明)
            for (var j = 0; j < positions.Length; j++)
            {
　　　//別の粒子モデルの位置
                var rock2 = positions[j];
           //粒子モデル間の距離計算
                float dist = math.distance(rock1, rock2);
          //粒子モデルが粒子間力発生範囲より小さい場合(すなわち粒子モデルが引力発生範囲に存在する時)
                if (dist < particleStickDistance)
                {
          //3次元成分から加わる方向ベクトルを算出
                    repulvector += rock1 - rock2;
                }
            }
          //正規化で方向ベクトルのみをリストに保存
            forces[index] = math.normalize(repulvector);
        }
    }
実際に力の大きさは数値的にユーザ側が決定するものであるため,定量的な判断が難しく,定性的な(見た目上の)判断に委ねられる。

このような解釈で合っていますでしょうか。

宜しくお願い致します

yosuke, 2024-07-19:
@MaruiYuta 
議論が噛み合っているかやや心配になってきているのですが、「数式で表現できるか」という一つ前の質問は、むしろ「理論式で表現ができたものを実装しており、最適パラメータを理論から導くことができるか」という意図なのではないかという気がしてきました。

その意図で正しい場合、答えとしては「土の挙動に関してはそもそも精密な理論式が存在しない」となります。
土は個体としての状態もあれば、粒子への破壊の過程もあれば、流体としての挙動や摩擦もあり、、、で、なかなかモデル化が大変な対象です。各シミュレータ開発者が手探りでそれらしい挙動を再現できる（＆実時間で動く実用性もある）モデルを作っているというのが現状です。

パラメータについては手探りで決定する必要があるのですが、ICPなどで自動の定量評価ができるようになれば、自動のパラメータ探索が可能になります。
その辺りの成果を今回は期待しています。

closed 2025-02-21 by maopotcha