pwri-opera/OperaSim-PhysX #12: 各種パラメータ調整をGUIで可能にする

maopotcha, 2023-04-07:
@yosuke 

以下の機能開発をお願いいたします。

### 概要
- 土砂や建機のパラメータをユーザがGUI上で簡単に調整できるようにする

### 目的
- 様々な作業対象物（土質）に、ユーザ側でシンプル対応可能な形にしたい

### やること
- 土砂（粒子）のサイズ/斥力/引力等に関するパラメータを調整するGUIを用意する
- 建機のアクチュエータ出力や挙動に関するパラメータを調整するGUIを用意する
- 該当するパラメータの選定についても、検討をお願いしたい
- 各パラメータの値域、意味、そのパラメータを変更することによって定性的な変化がどのようになるか等の情報をREADMEに追記する

### その他

yosuke, 2023-05-12:
バケットと土のインタラクションの設定については、バケットの先にスクリプトがアタッチされているが、深い場所にあるためわかりにくい＆設定項目が足りない。
[image]
粒子の大きさはPrefabで設定可能。これについてはREADMEに記述されていないことが問題と思われるので追記のみで良いかもしれない。
[image]

yosuke, 2023-05-12:
アクチュエータ出力の設定に関しては、以下で導入されたJointPosControllerでの対応が必要
https://github.com/pwri-opera/OperaSim-PhysX/commit/a911f1cb873a17dae0c1461bcc59bf02bac16285
FollowJointTrajectoryActionでも同様の処理があるので、そちらも忘れずに。

yosuke, 2023-05-12:
関節リミットやトルクリミットを規定するX Driveパラメータは、UnityEditorから各関節につき設定できる。
[image]
ただし、現在はJointPosControllerとFollowJointTrajectoryAction上で上書き設定している。
上書き設定部分を無効化してUnityEditor上からの設定を生かすようにするのが良さそう。
無効化とともに上記をREADMEに追記することで対応。

yosuke, 2023-05-12:
上記以外だと、PhysicsManagerから設定するDefault Solver Iterationsパラメータが非常に重要。
https://tech.pjin.jp/blog/2017/04/29/unity_physics_physics-manager/
既にUIで設定できるようになっているが、READMEに要追記。

yosuke, 2023-05-12:
こちらのブランチで作業中。
https://github.com/pwri-opera/OperaSim-PhysX/tree/feature/ui_based_joint_control_parameter
PosControlとTrajectoryActionが共存できるようにする変更も追加。

maopotcha, 2023-05-15:
ありがとうございます。
UIの形に拘る必要はありませんので、パラメータの調整方法と意味についての解説情報をREADMEに追記いただければ充分です。ある程度の開発単位が完了したら、小まめにPullRequestいただければ幸いです。

yosuke, 2023-06-23:
粒子シミュレーション部分のUI実装はとりあえず完了。READMEへの追記要。

maopotcha, 2024-02-08:
@yosuke 
先日、アップデートいただいたSoil Particle SettingsのGUIに表れている以下のパラメータについてREADMEに説明を追記いただけないでしょうか。
Angle of Repose
Repose Jitter
Sync Period
[image]

closed 2024-07-03 by maopotcha

maopotcha, 2024-07-03:
ツールチップに説明を追加したことで、本件クローズとします
https://github.com/pwri-opera/OperaSim-PhysX/issues/50