pwri-opera/OperaSim-PhysX #13: センサシミュレーション機能（センサモデル）

maopotcha, 2023-04-07:


yosuke, 2024-01-26:
GNSSに関して
- AWSIMについては、PoseWithCovarianceStamped型を使っておりNavsatfix型ではない
  https://github.com/tier4/AWSIM/blob/main/Assets/AWSIM/Scripts/Sensors/Gnss/GnssRos2Publisher.cs
- FRJのUnitySensorsについては、Navsatfix型で実装予定のようだが、今現在は実装が間に合っていない
   https://github.com/Field-Robotics-Japan/UnitySensors/blob/master/Assets/UnitySensorsROS/Runtime/Scripts/Publishers/GNSS/NavSatFixMsgPublisher.cs
難しい実装ではないので、この部分については内製してしまってもよさそう。

yosuke, 2024-01-26:
IMUに関して
- AWSIMについては、IMU型で実装あり。ただしROS2のみ対応なのでROS1もサポートしたい場合は実装を追加する必要がある。
   https://github.com/tier4/AWSIM/blob/main/Assets/AWSIM/Scripts/Sensors/Imu/ImuRos2Publisher.cs
- FRJのUnitySensorsについては、今現在は実装が間に合っていない（Publisher以外のセンサ値の生成部分はすでに実装されている）
   https://github.com/Field-Robotics-Japan/UnitySensors/blob/master/Assets/UnitySensorsROS/Runtime/Scripts/Publishers/IMU/IMUMsgPublisher.cs

yosuke, 2024-01-26:
ノイズに関して、AWSIM（の中で使われているRobotecAI Lidar）についてはガウスノイズを使ったシミュレーションあり
https://github.com/RobotecAI/RobotecGPULidar/blob/develop/docs/GaussianNoise.md

FRJ-UnitySensorsにもガウスノイズを使ったシミュレーションあり
https://github.com/Field-Robotics-Japan/UnitySensors/blob/master/Assets/UnitySensors/Runtime/Scripts/Utils/Noise/GaussianNoise.cs

両者ともノイズを加えているのはLiDARやDepthCameraに対してのみで、IMUやGNSSなどにはノイズを加えていないようなので、欲しい場合は実装を追加する必要あり。

yosuke, 2025-01-31:
@maopotcha 
LiDARは処理がそこそこ重いと思うので、スクリプトとしてはアタッチしつつも、チェックを外してデフォルトは無効（チェックを入れるだけで有効化できる）にしておくと良いかもしれませんね。
私の環境でも動かしてみます。

maopotcha, 2025-02-18:
https://github.com/pwri-opera/OperaSim-PhysX/pull/87
こちらに全て（GNSS, IMU, LiDAR, RGBカメラ）のセンサモデルを搭載し、LiDARとRGBカメラについてはデフォルトは無効にした状態にしました。こちらのPull Requestのマージをもって、本件完了にします。

closed 2025-03-12 by maopotcha