pwri-opera/OperaSim-PhysX #14: クローラモデルを修正する

maopotcha, 2023-04-07:
@yosuke 
以下の対応をお願いいたします。

### 概要
クローラ部に指令（/tracks/cmd_vel）を与えた時の挙動が実機と合わせ込みできるようにする

### 目的
- 同じ指令（/tracks/cmd_vel）に対し、実機とシミュレータとで走行部の滑りの挙動が異なる
　（おそらく、実機よりもシミュレータの方が旋回動作時の地面に対する滑りが大きくなっている）

### やること
- 同じ指令（/tracks/cmd_vel）に対する挙動が実機と会うようにパラメータを調整する
- 実機の平均値とシミュレーション結果とが合うようにしたい
- 滑りを決定するパラメータをユーザ側で調整可能なようにGUIを設けてもよい

### その他

yosuke, 2023-04-28:
Wheel Colliderを使うとForward FrictionとSideways Frictionを別個に設定できるので、特にクローラーの旋回滑りを調整する際に有効と思われる。
https://docs.unity3d.com/ja/2019.4/Manual/class-WheelCollider.html

yosuke, 2023-05-26:
各クローラーにWheel Colliderが設定してあるのだが、それと同時に元々のURDFで定義されているMeshColliderも有効になっているため、喧嘩してしまっていそう。
[image]

yosuke, 2023-07-28:
WheelColliderを有効にするには、親オブジェクトがRigidBodyである必要がある。URDFからインポートしたモデルはRigidBodyではなく、ArticulationBodyになってしまうので、今は設定されているものの有効化されていない状態。有効化されると緑の輪が出るので確認できる。
[image]

yosuke, 2023-07-28:
RigidBodyのリンクを最上位に作成して、その下にWheelCollliderを作成して、その下に既存のArticulationBodyをぶら下げればよいとのこと。ややハッキング的な解決法だが試してみる。
https://forum.unity.com/threads/articulationbody-and-wheel-colliders.1395376/#post-8784814

yosuke, 2023-07-28:
上記階層を作ればWheelCollliderを有効化できることを確認。

自動で作成されるサスペンションが原因で跳ね回ってしまうためパラメータ調整中。

yosuke, 2023-08-05:
クローラの挙動を再現するためにwheel colliderを縦に複数並べるとうまく回転できなくなるが、sideways stiffnessを調整することできれいに回転可能になった。
詳しくは以下のレポジトリ参照
https://github.com/brihernandez/PhysicsTankExample

yosuke, 2023-08-28:
wheel colliderが使われている直上のrigid bodyにcollision shapeが設定されていないと車両が安定しない。
https://qiita.com/comocc/items/5cc8423acb8d748112d1

unityは慣性行列(inertia tensor)の計算にcollision shapeを使っているらしく、collision shapeを設定しないと慣性行列がかなり小さくなってしまう。
さらにwheel colliderは直上のrigid bodyの慣性行列しか考慮してくれない（下にarticulation bodyをぶら下げても大きな慣性として考慮してくれない）ため、こうなるようだ。

適切な慣性行列を設定しても良いのだが、とりあえずはcollision shapeを設定することで対処する。

yosuke, 2023-08-28:
zx120についてはwheel colliderを使う形に切り替え完了。シミュレーションが破綻するような大きな飛び跳ねはなくなったが、微小に飛び跳ねているらしく、時間を置くと少しづつ位置がズレていってしまう。
[image]

yosuke, 2023-08-28:
本体のrigid bodyの慣性行列を大きめに設定したところ微小な振動が抑制されて、安定してきれいに動くようになった。

wheel colliderを使った方が旋回がきれいにできるようだ。

wheel colliderは、車軸への出力がvelocityではなくeffortなので、PIDコントローラなどで制御する必要あり。

yosuke, 2023-09-07:
PIDコントローラが実装できたので、ブランチを作成してpush完了。
とりあえずzx120のみを変更した状態だが、ダンプトラックについても動きを確かめたいので、もう少し作業してからmainブランチにマージ予定。
https://github.com/pwri-opera/OperaSim-PhysX/tree/feature/fix_wheel_colllision

yosuke, 2023-10-05:
diff drive controllerの単位変換部分が間違っていそうだったので修正を実施。

c30rの両前輪が回らないで踏ん張ってしまう問題あり。

上記問題以外は大体直せた状態。

maopotcha, 2024-02-08:
@yosuke 
クローラの旋回時の滑りを、ユーザ側で調整できるようにしていただきたいのですが、そのような調整パラメータをGUI上に設けて、READMEに使い方を追記いただけないでしょうか。

TaroABEJido, 2024-04-16:
@yosuke @maopotcha @genkiiii @numata-1740
下記の件について ，実機に於いて地面の状況が変われば滑り具合も変わりうると思うので，私もユーザ側で調整できるようにした方が良いと感じています．
https://github.com/pwri-opera/OperaSim-PhysX/issues/14#issuecomment-1933262145

また，このパラメータについてどの程度の範囲の値がよいか実機で調べておいて，何かしらのデータとして公開するのも良いとも思っています．

yosuke, 2024-04-18:
@maopotcha @TaroABEJido 
縦に複数の車輪が並んでいて、特に横滑りのパラメータについては車輪ごとに異なるものを設定している（旋回中心から遠いものほど横滑りしやすくしている）ので、その部分をもっと楽に設定できるようにできないか検討しています。
今のやり方だと、旋回速度がこれ以上「早くする」方向には調整できなさそうなのですが、調整したいのは路面状況に合わせて「遅くする」方向で良いでしょうか？
旋回を早くする調整が必要そうでしたらその方法も検討します。

yosuke, 2024-04-18:
@maopotcha @TaroABEJido 
あと、滑りやすさは路面のマテリアルとの相互作用なので、クローラーの滑り調整というよりはTerrainの滑りを調整する形でも検討しています（そちらのほうがパラメーター一つで調整できて楽かもしれません）。

yosuke, 2024-04-18:
Terrainの滑り係数（Physics Material）を変化させてみたが動作は変わらなかった。
Wheel colliderはPhysics Materialの設定を無視するとのこと。
https://docs.unity3d.com/Manual/wheel-colliders-friction.html

TaroABEJido, 2024-04-18:
@maopotcha，@numata-1740，@yosuke 
対応いただきありがとうございます．この件について遠藤さんとも相談してみます．

yosuke, 2024-04-18:
Diff Drive Controller上で、各車輪のForward FrictionとSideways Frictionを一括設定できるようにしてみたところ、こちらは良さそうな感じ。
c30rとic120ともに、Frictionを大きくしても軽い傾斜が登れない状態のため、PIDパラメータをもう少し調整した方が良い。

TaroABEJido, 2024-04-18:
@yosuke @maopotcha @numata-1740 
遠藤さんに相談したところ，
実機の挙動が滑りにより変わらないのではということ，
より厳密にするのであれば別のやり方の方が良いのでは
ということでした．

そのため，一旦この件は保留で良いとのことでした．
取り急ぎ連絡です．

yosuke, 2024-04-18:
@TaroABEJido @maopotcha 
とりあえず、ゆるい坂が登れない問題が発覚したので、そちらを直しています。
やや理論からは逸脱しますが、ForwardFrictionパラメータを「10」に設定するとクローラーが地面を掴んで踏ん張りが効くようになるようです。
あと車輪にクッション性を持たせると登るようになってくれそうなので、調整します。

yosuke, 2024-04-18:
Wheel Colliderのサスペンションのストロークは長めに設定した方が重機らしい動きになる（凹凸のある地面での踏ん張りも効くようになる）。残りの調整はまた後日。

kasahara-san, 2024-07-03:
本日のSIP会議での内容です。

本日の発表でご覧いただいた動画ですが、現在のmainブランチ(commit id：a78843c)に上がっているもののunity上のパラメータを独自に変更したものでした。
具体的には６つある"wheel_collider"の全ての"sideway_friction"の"extremum slip"の値を大きく(1000程度)した際に動画の状態になったとのことです
後輩に作業をしてもらったのですが、一部私の認識に齟齬があり、摩擦係数を大きくした際にこのような動作をしたとの誤認識のまま倉爪先生に伝えてしまっていました。申し訳ありません
変更内容として動画ように大幅に滑る状態になるのは、その通りかと思います

https://github.com/pwri-opera/OperaSim-PhysX/assets/130209264/ef04893b-e3ff-4a0b-a640-e7a08ebbffe9

ただ、添付画像のように、ic120停止時に横滑りする現象はデフォルトのパラメータ設定でも確認をしており、こちらについてご対応いただけますと幸いです。
よろしくお願いいたします
[image]

yosuke, 2024-07-03:
@kasahara-san (@maopotcha)
情報ありがとうございます。wheel colliderは色々なパラメータが設定できる分、どのパラメータをどう調整するのが良いか、こちらでも手探りの状態で、知見を共有し合いながら進められれば幸いです。

後半の横滑りの問題は、ある程度長く時間が経過するとこうなる、のだと理解しています。
wheel colliderとそれに固定された車体が微小に振動しており、それが様々な不具合の原因になっていることが、最近わかってきています。プロジェクトの物理シミュレーション設定で、 `Default Solver Iterations` を現在の50から最大の255に変更すると振動がかなり収まることもわかってきています。
https://docs.unity3d.com/ja/560/Manual/class-PhysicsManager.html

mainブランチの設定も変更しておきたいと思いますが、取り急ぎ情報共有させていただきます。

closed 2025-02-21 by maopotcha