pwri-opera/OperaSim-PhysX #18: 重機同士の干渉が無視されている

maopotcha, 2023-05-18:
各重機間の干渉が無視されているために、重機同士がぶつかってもすり抜けてしまう
各リンクオブジェクトのLayer設定が適切に行われていないのが原因と考えられる

Layerを整理して適切に設定することで対策する

yosuke, 2023-05-26:
ic120については、Default（他オブジェクト）との干渉が無効になっていますね。
[image]

yosuke, 2023-06-09:
zx200については、各リンクの干渉形状と実際のメッシュへの対応が無くなってしまっている状況。これだと干渉しない。
個々に再設定するのは大変なので、URDFからインポートし直すのが良いかもしれない。

[image]

yosuke, 2023-06-09:
c30rについても、vessel_link以外のリンクのcollisionが設定されていない（本来あるはずの部分が無くなっている）。

こちらも再インポートした方が良いかもしれない。

yosuke, 2023-06-09:
UnityにはGazeboの様な隣接リンク間の自己干渉をオフにする機能がない。

複雑なリンク構造の重機には隣接リンク間の自己干渉のオフは必須だが、レイヤで実現するとレイヤ構造が複座になりそうなため、下のURLのIgnoreCollision APIでプログラム的にオフにするのが良さそう。

https://docs.unity3d.com/ScriptReference/Physics.IgnoreCollision.html

レイヤとしては、できるだけシンプルにし

- 土砂の粒子
- バケット
- その他全て

の3レイヤ程度に整理したい。

yosuke, 2023-06-16:
以下のコミットでic120のコリジョンモデルを修正＆IgnoreCollisionAPIを使った形に変更。
https://github.com/pwri-opera/OperaSim-PhysX/commit/05bacaac772136078dcf2d2bea9e7a15f6c1a644

ic120とzx120についてはリアルに干渉するようになっている。
[image]

yosuke, 2023-09-22:
zx200を最新のURDF Importerを使ってインポートし直してみた。干渉形状は無事に復活。VHACDオプションを有効にしてインポートすると効率的に凸包形状に分割してくれるので、パフォーマンスも出そう。ただし、バケット部分の形状は少し変形してしまっているので、従来同様に手をかけてプリミティブ変換したほうが良いかもしれない。
[image]

yosuke, 2023-09-22:
ic120についてはIgnoreCollisionAPIを使って不要な干渉ペアを抑制したが、disable_collisionタグを使った方法も用意されているとのこと。
こちらの方がスマートなので、URDFデータに必要な設定を追加予定。
特にショベルカーの各関節のはめ合いはdisable_collision設定を追加しないと常に干渉した状態になっていると思われる。
https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/urdf_importer/urdf_appendix.md#disable-collision-support

yosuke, 2023-09-28:
ダンプトラックについては、URDFモデルの段階から干渉形状が設定されていない。
SA Mesh Colliderを使ってvisual形状からプリミティブを作ることで干渉形状とする。
[image]

yosuke, 2023-09-28:
干渉形状の修正は概ね完了。
まだwheel collider用のdiff drive controllerのパラメーターチューニングが甘いので、調整したのちプルリク作成予定。

yosuke, 2023-12-01:
@maopotcha 
#30 がマージされて解決したと思うので、issueを閉じます。
問題が再発した場合はreopenするか、新issueを立ててください。

closed 2023-12-01 by yosuke