pwri-opera/OperaSim-PhysX #19: Unityのバージョンアップ

genkiiii, 2023-05-18:
現状のUnityバージョンである2020.3.16f1では、Apple Siliconにネイティブ対応していません。そのためか、Mac上に起動したOperaSim-PhysXとROSのソフトが通信をした際にOperaSim-PhysXがフリーズすることがあります。
Apple　Siliconにネイティブ対応している2021.3.25f1ではフリーズ等の問題が発生しないので、リポジトリのUnityバージョンを2021.3.25f1（以上）にして欲しいです。

yosuke, 2023-05-19:
関連issue #16

yosuke, 2023-05-19:
新しい物理エンジンのDOTS Physicsが勝手に適用されないかが心配だったのですが、明示的に切り替えをしない限り大丈夫のようです。

常に最新のLTS版にするのが良いと思うので、動作確認の後、プルリク作成します。

yosuke, 2023-05-19:
@maopotcha @genkiiii 動作には問題なさそうでしたので、プルリク作りました。
https://github.com/pwri-opera/OperaSim-PhysX/pull/21

yosuke, 2023-06-23:
@maopotcha @genkiiii アップデート作業、動作検証含めてひと段落したと思うのでこのissueは閉じます（次のUnityバージョンも出ているようなので、追ってアップデートします）。

kimushun1101, 2023-11-09:
@yosuke @maopotcha @genkiiii 
Unity Editor のバージョンアップデートを確認いたしました．
しかしながら，それにともないREAMD.md と整合性が取れなくなっております．
https://github.com/pwri-opera/OperaSim-PhysX/blob/7fd0ed1a9f350b8a60505fe0446281615461f8fa/README.md?plain=1#L13-L14
今後もバージョン更新があるかと思いますので，README.md の文章をバージョンに依存しない構成に編集することをおすすめします．

yosuke, 2023-11-10:
マイナーバージョン無しで、Unity 2022 LTS、というような書き方が良いかもしれないですね。
https://gamemakers.jp/article/2023_06_01_40084/

yosuke, 2023-11-13:
プルリクエストも送っていただきありがたいです。 #32
CC: @maopotcha

maopotcha, 2024-02-08:
本件、一旦クローズにします。

closed 2024-02-08 by maopotcha