pwri-opera/OperaSim-PhysX #22: 実行直後にFPSが低下（処理負荷が過大）する

maopotcha, 2023-05-22:
概要
・実行直後にFPSが低下（処理負荷が過大）する
・実行直後にZX120のみ、作業機が落下（フリージョイント化？）してバケットがTerrainと干渉した結果、粒子計算負荷が増大するためと思われる
・取り急ぎ実行直後に作業機が落下しないようにすることで対処したい
・以下のpullリクをmerge後に生じるようになった模様

https://github.com/pwri-opera/OperaSim-PhysX/pull/21

yosuke, 2023-05-22:
あらら、私の方でも見てみます。 #20 の方で関節目標角設定のタイミングを変えたので、そちらの影響かもしれません。

maopotcha, 2023-05-22:
なるほど。早速のリプライありがとうございます。
（記憶では、マージ前の動作チェックではこのような現象は生じてなかったように思うのですが・・）

yosuke, 2023-05-22:
zx120とzx200を比較してわかったのですが、zx200の方はブーム以降の各リンクが異常に軽いmassが設定されていますね、、、。zx120の方も同様に軽いmassに設定したところ以前の動作に戻りました。

ただ、本来の重機のmassとはかけ離れているので、リアルなmassを設定した上で、関節制御のforce limitを大きく調整した方が良いかもしれません。zx120については、今はforceが重量に負けてブームが落ちてしまっている状態のようです。

@maopotcha

maopotcha, 2023-05-22:
@yosuke 
なるほど、そういうことでしたか。
この辺のパラメータは実機に合わせる作業がまったくできていない部分なので、まずは挙動がそれっぽくなるレベルで適当な数値を設定しておく、という対応にせざるを得ないと考えています。
メーカーさんに相談をしてみますが、きちんとした回答を得るのは少し時間がかかると思われます。

yosuke, 2023-05-26:
@maopotcha 
プルリク作成しました。#23

各リンクは「とりあえずの値」で250Kgに設定しておきました。

maopotcha, 2023-05-30:
@yosuke 
ご対応ありあがとうございます。本issue closeにします。

closed 2023-05-30 by maopotcha