pwri-opera/OperaSim-PhysX #31: rvizのmoveitパネルからバケット(とアーム)の操作ができなかった

MaruiYuta, 2023-11-10:
OperaSim-PhysXの概説5.ROSとの連携方法にて
"$ roslaunch zx120_unity zx120_standby.launch "まで実行しましたがrvizのmoveitパネルからバケット、アームの操作ができませんでした。
[image]

maopotcha, 2023-11-10:
@MaruiYuta 
ご連絡ありがとうございます。
こちらでも再現してみようとしたのですが、こちらの環境では（同様のERRORが出力されますが）今のところ問題なく動いています。
いくつか確認させてください。
【ROS側】
・UbuntuとROSのバージョンを教えてください
・Interactive Markerを動かした後に、MotionPlanning/Planning/Commands/Plan & Execiteは押しましたでしょうか
【Unity側】
・UnityはWindows版でしょうか、Linux版でしょうか
・TerrainにアタッチしているSoil Particle Settings（Script）のEnableのチェックを外してトライしてみてもらえますか
（地面の粒子化計算がOFFになります）
[image]

maopotcha, 2023-11-10:
あとはUnityとROSとが別のPC上で実行されている場合、2つのPC間の内部クロックが合致しているか（時刻同期とれているか）確認していただけますでしょうか

MaruiYuta, 2023-11-28:
返信が遅くなってしまい大変申し訳ございません。

> UbuntuとROSのバージョンを教えてください
ROS：noetic
Ubuntu：Ubuntu 20.04.6 LTS \n \l

> Interactive Markerを動かした後に、MotionPlanning/Planning/Commands/Plan & Executeは押しましたでしょうか
はい、押しました。unity上の油圧ショベルの動作を確認しましたが、動いていない状況です。

> UnityはWindows版でしょうか、Linux版でしょうか
Linax版で行っています

> TerrainにアタッチしているSoil Particle Settings（Script）のEnableのチェックを外してトライしてみてもらえますか
（地面の粒子化計算がOFFになります）
[image]
実行しましたが動作が確認できませんでした。

> あとはUnityとROSとが別のPC上で実行されている場合、2つのPC間の内部クロックが合致しているか（時刻同期とれているか）確認していただけますでしょうか
１つのPCで起動しているため、問題ないかと思います

宜しくお願い致します

yosuke, 2023-12-01:
@MaruiYuta @maopotcha 
以前経験したことがある現象ですが、同じPC上で動かしていても、そのPCに複数のネットワークインターフェースが存在する場合（「有線LANとWifi」や、「2つの有線LAN」など）に、ROSの内部通信が疎通しないことがありました。
その時には環境変数 `ROS_IP` を設定することで解決した覚えがあります。
このissueについては違う原因かもしれませんが、可能性の一つとして情報提供させていただきます。

MaruiYuta, 2023-12-01:
@yosuke @maopotcha 
回答していただきありがとうございます。
先程指導教員と相談した結果、今後土木研究所で行った結果をROS bagで持ち帰り、自分の研究室で作業する機会もあるだろうから同じような環境づくりを行ったほうが良いというアドバイスをいただきました。

よってそちらと同じ環境を整えようと考えております。
念の為、そちらの環境を教えていただければ幸いです。

また現在、UnitySensorを使用して3次元LiDARを自身が簡単に作ったプロジェクト内で動かし、実装できることを確認した段階です。環境のセットアップができ次第、UnitySensorの実装、データの取得という流れで進めていく予定です。

宜しくお願い致します

MaruiYuta, 2023-12-09:
@maopotcha 
回答ありがとうございます。
現在も環境セットアップ段階です。進捗が遅く申し訳ありません。
幾つか分からない部分があったため質問させていただきます。
以下に記載します。
質問事項
(1)PCは2台、OSはwindows,Linaxが入ったものを用意する必要があると思うのですが正しいでしょうか。（その場合、機材がそろっているため可能です）
(2)OSのバージョンが分かれば教えていただきたいです。
(3)(1)が正しい場合、windowsPC,LinaxPCに必要なソフトウェアを教えていただきたいです。

以上になります。

現状デスクトップパソコンを2台使用し、windows,Linax環境を整えようと思っています。
返信が遅くなってしまい申し訳ございません。
回答して頂けたら幸いです。よろしくお願いいたします。

maopotcha, 2023-12-10:
（１）正しいです。可能であれば、有線ＬＡＮで接続可能なものが望ましいです（無線通信の問題と切り分けるため）
（２）Windowsは10, LinuxはUnutu18.04（ROS1の場合)又はUbuntu22.04（ROS2の場合）になります。Windowsの方は11でも大丈夫だと思います
（３）特にありません。WindowsにはUnityとGit系の何かしらのバージョン管理ツール、Linux側はros-melodic（ROS1）またはros-humble（ROS2）をインストールして、各パッケージのREADMEに記載の手順で進めていただければ、ひとまずROSからUnity内の建機を動かせる状態になると思います。

- https://github.com/pwri-opera/OperaSim-PhysX
- https://github.com/pwri-opera/zx120_ros
- https://github.com/pwri-opera/zx120_ros2

MaruiYuta, 2023-12-10:
@maopotcha 
回答ありがとうございます。
環境について記載します。問題がある場合、教えていただけたら幸いです。
ノートPCに既にros-noeticが入っているためdockerを使用しros-melodic(ROS1)を導入、デスクトップPCにwindows10または11を導入します。ソフトウェアは教えていただいたものを用意します。

自分が経験したことがないため質問したいのですが、Linax上のROS1とwindows上のシュミレータ(Unity)への通信はどのように行っているのでしょうか。

maopotcha, 2023-12-11:
@MaruiYuta 
https://user-images.githubusercontent.com/24404939/161001271-0f81d211-4c8e-4341-8f9f-86a02e958c4d.jpg
LinuxとWindowの通信ですが、基本的には上記の図の通りです。
ROS-TCP-Endpoint/Connectorという通信モジュールを使ってやり取りします。
ハードウェアとしてはLANケーブル又はWiFiで繋がる形であれば通信は可能ですが、まずは有線のLANケーブルで繋げることをお勧めします。
dockerはネットワーク設定が少々面倒かと思いますが、頑張ってください。
PCをUbuntu18.04と20.04のデュアルブートにしてしまう、というのも手です。

MaruiYuta, 2023-12-12:
@maopotcha 
有難うございます。
時間がかかるため2台、デスクトップPCを研究用に用意し環境整備することにしました。現在、使用するデスクトップPCのスペックに悩んでいます。以前windows版GPU,CPUの情報を教えていただいたのでそのようなPCを用意しようと思っています。可能であればlinax版はどのようなスペックなのか教えていただきたいです。

MaruiYuta, 2023-12-21:
@maopotcha 
教えていただいた環境を整え、シュミレーションができるようになりました。ありがとうございます。
motionPlanningを使用した実行でなく、特定の動きをするようノードを作成したいと考えております。rqtを確認すると/zx120/robot_state_publisherに/zx120/joint_statesを流しているのは/unity_endpointであるためこのノードを編集すればよいと考えました。まだ理解できていない部分が多いですが、どこを見ればノード作成ができ、モデルに特定の掘削を行わせられるのでしょうか。

maopotcha, 2023-12-25:
@MaruiYuta 
endpointはUnity側とトピックの送受信をする既存ノードなので手を加えない方が良いかと思います。
ノードの作成方法についてはROSのTutorialを参照されるのが良いです。（https://wiki.ros.org/ja/ROS/Tutorials）
あとはhttps://github.com/pwri-opera/zx120_rosのReadMeを元に動かせる状況にあるのであれば
MoveIt!のインターフェースを用いて動かすか、或いはendpointが持っている関節動作指令を直接作る、のいずれかを実行するパッケージの自作を検討いただくのが良いと思います。

maopotcha, 2024-02-08:
本件、お手元の環境で動作したようなので一旦クローズにします

closed 2024-02-08 by maopotcha