pwri-opera/OperaSim-PhysX #38: シミュレータ/ROS間の通信IFを共通制御信号(com3_ros）に合致させる

maopotcha, 2024-02-09:
### 概要
シミュレータ/ROS間の通信インターフェースを、実建機の変更に合わせるように修正する

### 目的
共通制御信号の改定に合わせて、シミュレータ内の建機を動作させるための通信IOを修正する

### やること
- 共通制御信号の改定案が決定した後に、この仕様に合わせてUnityシミュレータ側の送受信トピックの名称および型を変更する

### その他
- 共通制御信号の改定は24年3月頃の見込み
- 対応順序はzx200, ic120, zx120の順番となる見込み

yosuke, 2024-07-09:
com3はCANベースの定義になっており、メッセージのシリアライズとデシリアライズにdbc形式の定義ファイルを使う。

油圧ショベル用のメッセージ定義は以下で行われている。

https://github.com/pwri-opera/com3/blob/main/excavator_com3/excavator_com3.dbc

dbc形式の詳細は以下を参照。CANでよく使われる定義ファイルだが、標準から改変なく利用されている。

https://docs.fileformat.com/ja/database/dbc/

yosuke, 2024-07-09:
CANメッセージのデコードとROSメッセージへの変換とpublishを行うブリッジプログラムが以下で実装されている。

https://github.com/pwri-opera/zx200_com3_ros/blob/main/include/zx200_com3_ros.hpp

変換される先のROSメッセージのデータ型定義；

https://github.com/pwri-opera/com3_ros/tree/main/com3_msgs/msg

zx200用に利用されているトピック名一覧：

https://github.com/pwri-opera/com3_ros/blob/main/excavator_com3_ros/zx200_cmd_list.md

yosuke, 2024-07-10:
dbc定義ファイルを読み込んでメッセージのシリアライズとデシリアライズができるC#用のライブラリもあるので、Unity上でCANメッセージを直接扱うことも可能。

https://github.com/EFeru/DbcParser

ただし、まずは一つ前のROSブリッジを使った形の対応をするのが正しいと思われる。

yosuke, 2024-07-10:
ROSブリッジを使う際に対応が必要なのは主に以下のトピック

- /(excavator_name)/front_cmd アームの各関節コマンド
- /(excavator_name)/tracks_cmd クローラコマンド

クローラコマンドについては、cmd_velも生かしておきたいので、両立できるアルゴリズムを実装する方が良い。

以下のトピックについては、シミュレーションできていない内容が多いので、現時点ではスキップするか、ダミー値を入れる実装を行う。

- /(excavator_name)/machine_setting

yosuke, 2024-07-10:
ROSブリッジ上では、トルク(effort)ベースのコマンドは現時点では実装されていない。

https://github.com/pwri-opera/zx200_com3_ros/blob/main/include/zx200_com3_ros.hpp#L332-L347

- positionのみが設定された場合は、最大速度で目標角まで移動
- velocityのみが設定された場合は、指定速度で回転（クローラコマンドなどはほとんどこの使い方）

とするのが良いと思われる。

- positionとvelocityの両方が設定された場合は、指定速度で目標角まで移動

となると思うが、目標角到達の誤差を減らすには、目標前で速度を落とした方が良いと思われる。
細かい挙動の実装については、要確認。

yosuke, 2024-07-10:
tracks_cmdに関して、

zx200については、left_trackとright_trackとなっているが、
https://github.com/pwri-opera/zx200_com3_ros/blob/main/include/zx200_com3_ros.hpp#L411-L432

ic120については、forward_volumeとturn_volumeとなっている、
https://github.com/pwri-opera/ic120_com3_ros/blob/main/include/ic120_com3_ros.hpp#L109-L129

揃えた方が良いか、この程度の差は許容するか要確認。

genkiiii, 2024-07-10:
> ROSブリッジ上では、トルク(effort)ベースのコマンドは現時点では実装されていない。
> 
> https://github.com/pwri-opera/zx200_com3_ros/blob/main/include/zx200_com3_ros.hpp#L332-L347
> 
> * positionのみが設定された場合は、最大速度で目標角まで移動
> * velocityのみが設定された場合は、指定速度で回転（クローラコマンドなどはほとんどこの使い方）
> 
> とするのが良いと思われる。
> 
> * positionとvelocityの両方が設定された場合は、指定速度で目標角まで移動
> 
> となると思うが、目標角到達の誤差を減らすには、目標前で速度を落とした方が良いと思われる（その場合は、「指定された最大速度で目標角まで移動」となる）。 細かい挙動の実装については、要確認。

現状では、シミュレータと外部との通信は、ROS-TCP-Endpointに依存しており、com3以外の通信(Lidarやカメラ等)もやりとりされています。そのため、com3関係の通信のみ擬似的にCANベースのものを利用せずに、すべて通常のros msgにしたいと思います。

また、現状は、制御モード(positionとvelocity、effort)は排他的に利用する形としています。
まずは、シミュレータのインスペクタにて、制御モードを選択できるようにしていただけますでしょうか。
AGX版の方はそのような実装になっています。

genkiiii, 2024-07-10:
> tracks_cmdに関して、
>
> zx200については、left_trackとright_trackとなっているが、 https://github.com/pwri-opera/zx200_com3_ros/blob/main/include/zx200_com3_ros.hpp#L411-L432
>
> ic120については、forward_volumeとturn_volumeとなっている、 [...]

こちらは実機の操作方法に依存する形の実装になっています。
油圧ショベルのクローラ操作は、左右のクローラを２つのレバーで動かしているため、左右クローラの動きをそれぞれ指定しています。
一方で、クローラダンプのクローラ操作は、２つのレバーと動かすパターンと１レバーで動かすパターンがあります。
ic120は後者の１レバーパターンのため、このような実装になっています。

yosuke, 2024-07-10:
@genkiiii 諸々了解しました。そのように実装します。

yosuke, 2024-07-10:
とりあえずROSのcom3メッセージ定義をインポート完了。
ROS1, ROS2のどちらでも問題なくスタブ生成できていそう。

yosuke, 2024-07-12:
front_cmdの実装をとりあえず作成完了（テストはこれから）。

track_cmdの方はDiffDriveController内に実装してして、left_trackとright_trackのパターンと、forward_volumeとturn_volumeのパターンの、どちらのコマンドも受けられるようにしてしまうのが良いか。

yosuke, 2024-07-17:
tracks_cmdについても実装完了（テストはこれから）。

yosuke, 2024-07-17:
簡易なテストコマンドが下のURLにあり。
https://github.com/pwri-opera/com3_ros/blob/main/sample_ros_cmd.sh

yosuke, 2024-07-26:
`/(excavator_name)/emg_stop` トピックも重要機能なので要実装。com3用のトピックだが、既存のROSトピックについても非常停止するのが良いと思う。

closed 2025-02-21 by maopotcha