pwri-opera/OperaSim-PhysX #44: シミュレータの関節挙動を実機に近づける

TaroABEJido, 2024-04-16:
## 概要
シミュレータの関節挙動を実機に近づける

## 目的
現状，シミュレータ内建機の関節には，実機に発生するような操作指令が発生した場合のむだ時間等の要素が考慮されていない．より実機に近いシミュレータにする場合にはその点も考慮する必要がある．

## その他
本件すでに課題点としてチーム内で共有されているかと思うのですが，シミュレータの改善点の一つなので issue として投稿しておきます

TaroABEJido, 2024-04-17:
## 参考になりそうな文献一覧
見つけ次第随時更新
(例)[文献名] : [説明]

*

maopotcha, 2024-05-01:
@TaroABEJido 
むだ時間含め、cmdを受信してからの建機の応答特性については全てUnity側に実装をお願いします。

maopotcha, 2024-05-01:
ひとまず油圧ショベルについては「プレ大会仕様の完成」のマイルストーンまでに完了を目標にしましょう

TaroABEJido, 2024-05-01:
> @TaroABEJido むだ時間含め、cmdを受信してからの建機の応答特性については全てUnity側に実装をお願いします。
承知しました

TaroABEJido, 2024-05-01:
承知しました

TaroABEJido, 2024-08-09:
## zx120 関節パラメータの選定

### 概要
各部位の質量を実機に合せ，カタログ値などを参考に Force limit を設定したうえで，XDrive 内 stiffness，damping それぞれの組み合わせを試し，実機の挙動とシミュレーションの挙動が近くなるようなパラメータを選定する． 

### 内容
- boom, arm, bucket, swing について，stiffness, damping の組み合わせを 25 通り施行する
    - Force Limit をそれぞれ以下のように設定
        - swing: 3200000 [Nm] ← 以下関節の設定値値から推測 
        - boom: 400000 [Nm]   ← 以下のarm, bucket の設定値から推測
        - arm: 236187 [Nm]        ← カタログ最大掘削力より推定
        - bucket: 121770 [Nm]   ← カタログ最大掘削力より推定

    - 各部位の重量は以下のように設定
        - クローラ部: 3816 [kg]
        - 上部旋回体: 6514 [kg]
        - ブーム: 992 [kg]
        - アーム: 392 [kg] 
        - バケット: 386 [kg]

    - Swing に関しては 一往復分ステップ応答を与えたときの，実機関節角とUnity 内機体関節角度の二乗誤差が小さくなるものを選定
    - Boom, Arm, Bucket に関しては二往復分のステップ応答を与えたときの二乗誤差が小さくなるものを選定
    - ※ Stiffness, damping の値が小さすぎるとトルク不足になる傾向があるため，誤差が小さくなるパラメータの内，値が大きめのものを選定

    - パラメーラ選定結果は ブランチ /feature/param_set に 反映

- 実験結果
    - Swing
        [image]

    - Boom
        [image]

    - Arm
        [image]

    - Bucket
        [image]

- 選定結果
    - Swing
        - Stiffness: 2000000
        - Damping: 2000000
    - Boom
        - Stiffness: 1000000
        - Damping: 2000000
    - Arm
        - Stiffness: 2000000
        - Damping: 2000000
    - Bucket
        - Stiffness: 1000000
        - Damping: 2000000

TaroABEJido, 2024-08-22:
## zx120 関節パラメータの微調整

### 概要
以前コメントした  [zx120 関節のパラメータ](https://github.com/pwri-opera/OperaSim-PhysX/issues/44#issuecomment-2276926331) の設定 について，boom と bucket の応答が実機と異なる印象がある．設定されたパラメータを基準に，目標角度への収束時間や見た目を参考にして微調整を行う．

### 内容
* boom の関節パラメータ微調整
    * Dumping の値を 調整する．
  
 [image]

    |  |Dumping: 1000000|
    | ---- | ---- |
    | [image] |[image]  |

    |  |  |
    | ---- | ---- |
    |  |  |

* bucket

TaroABEJido, 2024-10-18:
## ic120 vessel 関節パラメータ調整
### 概要
- 物理シミュレータ内 IC120 ベッセルの応答合わせ
    - 物理シミュレータ内 ic120 ベッセル角度変化を，実機における ベッセル上げ下げ時のベッセル角度変化とできるだけマッチするように調整する．
   
### 内容
* feature/set_param ブランチに 角度変化を調整したものをpushした．
  * https://github.com/pwri-opera/OperaSim-PhysX/tree/feature/ic120_setting 
* 調整後の関節挙動
    - データ詳細
        - 縦軸横軸
            - 縦軸：角度[rad]
            - 横軸：時間[sec]
        - 凡例
            - 赤線：物理シミュレータにおけるベッセル角度変化
            - 青線：ベッセル角度指令 (シミュレータへの関節角度入力用)，
                - ※実機とシミュレータに於いて，入力の与え方に差異あり
                    - 実機: ベッセルの上げ下げをレバー入力最大で実施，稼働範囲上限までベッセル上げ -> 元の位置に戻す．
                    - ※シミュレータ: 実機に於いて操作入力が始まったタイミングで位置指令を出力
                        - ベッセル上げ -> -65 deg (約1.13 rad) を送信
                        - ベッセル下げ -> 0 deg を送信
            - 水色：実機におけるベッセル角度変化

[image]

※実機とシミュレータに於いて，入力の与え方に差異あり

- 実機: ベッセルの上げ下げをレバー入力最大で実施，稼働範囲上限までベッセル上げ -> 元の位置に戻す．
- シミュレータ: 実機に於いて操作入力が始まったタイミングで位置指令を出力
    - ベッセル上げ -> -65 deg (約1.13 rad) を送信
    - ベッセル下げ -> 0 deg を送信

TaroABEJido, 2024-10-18:
ic120 ベッセル駆動については，将来的には 角度を入力 -> 内部的には targetVelocity を与える形にした方がよさそう