pwri-opera/OperaSim-PhysX #45: 建設DXフィールドの航空測量データから作成した3DマップをTerrainに取り込む

maopotcha, 2024-04-19:
概要
建設DXフィールドの地形データをTerrainに反映できるようにする

目的

建設DXフィールドにおける動作の事前トライアルができるようにする

やること
建設DXフィールドの地形データをTerrainとして取り込む
建設DXフィールドのTerrainをデフォルトにする
READMEに建設DXフィールド（〇●年△月×日時点）の航空測量データである旨、追記する

yosuke, 2024-04-26:
以前同様の取り込みをした際には、元データはLAS形式であり、下記のツールに解像度を調整するパッチを当てたものを使ってheightmapに変換し、それをTerrainに反映していた。
https://github.com/jhawthorn/las2heightmap

yosuke, 2024-04-26:
レーザー計測（LAS形式）ではない、写真計測（SfM）に関しても最近はツールが充実している。
以下のチュートリアルの後半にSfMで地形モデルを作成し、heightfield（heightmap）を生成するまでの手順が解説されている。
http://topography.csis.u-tokyo.ac.jp/resources/171029_gisa_sfm/SfM-MVS_photogrammetry_basic_v2.pdf

maopotcha, 2024-04-26:
@yosuke 
ありがとうございます。
取り急ぎ、以前に作成いただいたDX実験フィールドの航空写真測量データをTerrainに設定してみていただけますでしょうか。
建機の動作などに問題なければ、一旦はデフォルトTerrainをこちらにしてしまおうかと考えております。

yosuke, 2024-05-10:
@maopotcha 
了解です。
DXフィールドのTerrainはかなり大きく、前回設定した時にはTerrainを格子状に分割して、重機の周りだけ有効化することであまり遅くならないようにするなどしていたのですが、もしかしたら最近有効化した `Broadphase Type` の設定で、格子への分割と同じ効果があるかもしれないので、まずはシンプルな形でやってみます。

yosuke, 2024-05-24:
とりあえずheightmapをインポートしてみた（textureについてはまだ）。
最高解像度の設定でインポートするとやはりかなり遅くなる（fps 1前後）ようなので、要調整。
[image]

yosuke, 2024-05-24:
textureこみのインポートについても無事に成功。textureが付くとかなりリアルになる。
残るはパフォーマンスの問題。
[image]

yosuke, 2024-05-29:
安息角のcompute shader計算を無効化したところ、fpsが大きく改善した。毎フレームに実行しないなど工夫をすると良いかもしれない。

compute shader計算が有効な場合も、heightmap resolutionを下げるとfpsが大きく改善する。
1025 x 1025まで下げると120 fps程度の実用的な値になった。

yosuke, 2024-05-29:
heightmap resolutionが最高解像度の4097 x 4097だと遅い理由は、開発に使っているRTX3060ビデオカードのCUDAコア数が3,584基で4097より少ないためと思われる。

yosuke, 2024-05-31:
解像度が高い状態でもcomputer shaderの並列度設定を調整するとfpsは100を超える。並列度の調整後もきちんと計算できているかを確認後に再度採用を検討。
解像度が高いとheightmapをterrainに反映させるsync処理がかなり遅くなるので、最初の一回のみ適用する（以降は建機の周囲のみsyncする）処理に変更するのが良いと思われる。

yosuke, 2024-06-05:
syncの頻度を調整すると実用的な速度になるが、地面を掘る局面では頻繁なsyncが必須になる。terrain全体ではなくバケットの周辺だけのsyncをするように実装しているはずなのだが、動作がかなり遅い。
syncの挙動については詳細を調査したほうが良いとかねてから思っていたので、この機会に調査を実施。
ソースコードは下のURLから閲覧できる。
https://github.com/Unity-Technologies/UnityCsReference/blob/master/Modules/Terrain/Public/TerrainData.GPUCopy.cs

yosuke, 2024-06-05:
sync関係のAPIの呼び出しにcallbackを登録できる仕組みになっており、変なcallbackが登録されていると遅くなる可能性があるが、比較的新しい機能のようなので使っている依存プラグインは少なさそう。
https://github.com/Unity-Technologies/UnityCsReference/blob/b44c4cc9e4ce3dfa0bab2fe4bf7efae880c5a175/Modules/Terrain/Public/TerrainData.GPUCopy.cs#L152

syncの実体部分はネイティブコードで実装されており中を見ることができない。
unity純正のterrainの利用を諦めて自分で実装する人も多いらしい（AGXUnityも自分で実装したterrainクラスを使っている）が、もう少し調査を継続。

yosuke, 2024-06-12:
`Terrain.RecomputeInvalidPatches`は、heightmapの解像度がある程度大きくなると急に大きく遅くなるとのこと。
https://forum.unity.com/threads/syncheightmaps-and-applydelayedheightmapmodification-performance.658675/#post-4422712
CPUからGPUに転送する際のバス幅の問題であろうとのこと。
terrainを小さなグリッドに切り分けるのが有効な対策。

yosuke, 2024-06-12:
terrainのグリッド分割、以前はサードパーティーツールを使ったのだが、今は公式のツールがあるとのこと。
https://qiita.com/Cova8bitdot/items/4f465672886728a9f6ad

yosuke, 2024-06-12:
公式ツールによるグリッド分割、heightmapは綺麗に分割してくれるが、テクスチャは分割してくれないので別途対処の必要あり。

yosuke, 2024-06-19:
Unity公式のterrain toolはグリッド分割されたterrainに対するペイントや変形操作もできる、という情報も見つけたのだが、実装を見てみるとペイントや変形は一部機能しか対応していないようで、無い部分は自分で作る必要がある。

テクスチャは、terrainタイル毎にオフセットを設定するとグリッド分割されていても適切に表示できる。タイル数が多いと手動で設定するのが大変なので、簡易な設定ツールを作るのが良いと思う。

一番大きな問題は、掘削時に地面の変形が起こった際のheightmapの管理。heightmapのterrainへのsyncが遅いだけで、heightmap自体はワールド全体で1枚でも問題ないため、安息角を使った地面の崩壊の計算などを単純化するために、マスタのheightmapを1枚にして持っておくのが良いと思う。
heightmapをテクスチャ・アセットとして保存しておき、terrainタイルと別途ロードするのが良いか。
https://qiita.com/k7a/items/df6dd8ea66cbc5a1e21d#resourcesload

yosuke, 2024-06-19:
今後、DXフィールド以外のワールドも使いたい、というリクエストもありそうで、ユーザ側でタイル分割などのオペレーションをするのは大変なので、terrainのデータは1枚で持っておいて、シミュレーション開始時に動的に分割する方が良いかもしれない。

heightmapのマップタイルへの分割は以下の部分で実装されており、かなり単純な処理。
https://github.com/needle-mirror/com.unity.terrain-tools/blob/71917f5adb362efcc699668415a0516f732cef71/Editor/TerrainToolbox/TerrainToolboxCreateTerrain.cs#L997-L1006

yosuke, 2024-06-26:
heightmapをフル解像度で作成するとassetファイルが100MBを超えるためLFSを使わないとgithubにpushできない。
ユーザ側でのLFSの導入には若干の手数がかかり、敷居の高さにもつながるので、データサイズを小さくできないか試行中（サイズが100MBを少しだけ超えている状況）。

yosuke, 2024-06-26:
control textureの解像度を小さくしたところassetのサイズが小さくなってくれた。control textureはマテリアルの挙動の解像度に関わるが、今回はマテリアルについては複雑なことをしていないので、問題ないはず。

yosuke, 2024-07-03:
terrainの自動分割処理のスクリプト実装完了。かなり早くなって良いのだが、分割後のマップタイル間に筋ができてしまう問題があり修正中。

[image]

yosuke, 2024-09-13:
マップタイルの隙間の問題、分割計算時の小数点誤差の問題だったようで、無事に解決できた。

あとは、BucketRocksの部分をタイル化されたTerrainに対応させる必要あり。

yosuke, 2024-09-20:
BucketRocksのタイル化されたTerrainへの対応も、どうにか動き始めてくれた。

座標のズレがあるので、要デバッグ。

yosuke, 2024-09-20:
筋ができる件の対処が座標のズレを生じさせている。

筋の方を綺麗に解決したいのだが、まだ解決法が見つかっていない状況。以下のURLと同じ問題。
https://discussions.unity.com/t/gaps-between-terrain-tiles/768823

yosuke, 2024-09-27:
筋の件、無事に解決完了。タイル化された各Terrainの境界のheightを厳密に一致させないと接続してくれないUnityの仕様が原因。

細かな調整をして完成予定。

yosuke, 2024-10-11:
安息角の計算についてはタイル化は行わず、1枚のheightmapに対して計算している（境界の扱いをシンプル化するため）が、広いフィールドだと計算量が大きくなる。

本来は建機の周辺だけ計算してアップデートすれば良いはずなので、この部分も改良したい。

昨今のコンピュータに搭載されたGPUメモリ量を考えると、1枚のheightmapで扱うこと自体は無理はないと考えられる。