pwri-opera/OperaSim-PhysX #46: 文書化システムの導入を検討する

yosuke, 2024-04-26:
### 概要
文書を充実していきたいが、READMEに書く形だと長い文章を書くことができない。
章立で整理できる文書化システムを導入したい。

### 目的
読みやすく詳細に記述された文書を作成する。

### やること
- wikiやSPHINXなど複数の候補からメリット・デメリットを整理し適切な文書化システムを選択する。

yosuke, 2024-04-26:
Githubのwikiを使った方法

### メリット

- 更新が楽にできる

### デメリット

- Githubは検索エンジンからのインデキシングを制限しており、検索エンジンに引っかかりにくい
- subsectionなど細かい章立てにした文書を書きにくい

yosuke, 2024-04-26:
SPHINXを使った方法

SPHINXは元はPython用に書かれた文書化システムであったが、今は一般の文書化システムとして広く用いられている。

有名なものではOpenCVの文書もSPHINX形式：
https://docs.opencv.org/2.4.13.7/

その他多くの採用事例がある：
https://www.sphinx-doc.org/ja/master/examples.html

### メリット
- subsectionを含んだ章立ての文書を作成できる
- 作成した文書をHTMLだけでなくPDF等でも出力できる
- 数式にも対応
- 検索エンジンにインデキシングされやすい任意のサイトでホストできる
- SPHINX形式の文書を無料でホストできるreadTheDocsサービスもある https://about.readthedocs.com/

### デメリット
- 簡易なテキスト形式で文書を書くことができるが、多少の学習が必要

numata-1740, 2024-05-29:
@yosuke 様
@numata-1740 です。
お忙しいところ恐縮ですが
Sphinx のインストールが完了しました。
ローカルにも、作成されたHTML形式ファイルを準備できました。
よろしくお願いいたします。

closed 2024-07-03 by maopotcha