pwri-opera/OperaSim-PhysX #47: IMUおよびGNSSセンサを追加する

yosuke, 2024-04-26:
#13 のサブタスク

### 概要
- 自動化建設機械に用いられる代表的なセンサについて、シミュレータ側から模擬出力をROSメッセージとして出力する機能を付加する

### 目的
- 自律施工ソフトウェアを開発するために、センサシミュレーション機能を追加する
- 手始めとしてIMUとGNSSセンサを追加する

### やること
- GNSSシミュレーション機能 : [Navsatfix型](https://docs.ros2.org/foxy/api/sensor_msgs/msg/NavSatFix.html)
- IMUシミュレーション機能 : [Imu型](https://docs.ros2.org/foxy/api/sensor_msgs/msg/Imu.html)
- PhysX版シミュレータの4台の重機にそれぞれのセンサを追加する
- センサの追加箇所は実機の位置となるべく一致させる

numata-1740, 2024-04-26:
ご連絡ありがとうございます

numata-1740, 2024-04-26:
ご連絡ありがとうございます。
本日、午前中は、昨日の筑波大実験の続きのため
対応できませんでした。
ご依頼のありましたGitHub アカウントをご連絡いたします。
よろしくお願いいたします。

numata-1740, 2024-04-26:
[image]

numata-1740, 2024-04-26:
[image]
昨日、IMUの設置場所を調べた写真です。zx120　ショベルです

numata-1740, 2024-04-26:
１）Backet部

numata-1740, 2024-04-26:
Backet部　
[image]
Arm部
[image]
Boom部
[image]
Base部：操作室横
[image]

numata-1740, 2024-04-26:
IMUセンサの型式を調査して、加速度センサの内容を調べる予定です。
設定場所の詳細な位置まで必要でしょうか？
まだ十分な状況を把握していませんので、認識がずれているかもしれませんが
教えてください。よろしくお願いいたします

TaroABEJido, 2024-04-30:
# zx200 の IMU 配置
zx200 の imu について，配置場所の写真をアップロードします
|[image]|[image]|[image]|
|---|---|---|

TaroABEJido, 2024-04-30:
@yosuke 
zx200 について 一通りGNSSとIMUを追加しました
https://github.com/pwri-opera/OperaSim-PhysX/tree/6ed4392ffcceaffafdc51c6acdcdd5fc850aeba1．

以下が主な変更点です．
* body 上に gnss を二基追加
  * GeoCooodinateSysytem もプロジェクトに追加（座標は土研のDXフィールド）
* bucket, arm, boom, body にそれぞれIMU を追加

IMUの正確な配置（特に向き）は分からないので，見た目で適当に配置しています．

TaroABEJido, 2024-05-02:
# バケットのIMUに関する扱い方
* バケットの IMU に関して 実機に於いてはバケットのリンク？を構成する部品に対して IMU が搭載されているが，現状 Unity 内ショベルには下図のように該当のリンクが存在しないため，実機のIMU をどのように物理シミュレータで再現するかという問題がある．
[image]

* 実機においては，上記図該当部に設置されたIMUの情報からバケットの関節角度等を幾何学的情報に基づいて算出，Topic として出力している．
* Unity 内ショベルに於いては，バケットリンクにIMUを定義することは可能であるため，幾何学的情報さえあれば実機におけるIMUより計測される姿勢情報は再現可能であるが，加速度や角速度の情報の再現が困難である．

yosuke, 2024-05-07:
@numata-1740 

- センサの追加箇所は実機の位置と***なるべく***一致させる

ということなので、位置はそこまで厳密でなくて良いです。

yosuke, 2024-05-07:
@numata-1740 
IMUの型番がわかったら、「データの出力レート」についても可能であれば合わせてください。

yosuke, 2024-05-07:
@numata-1740 
作業が進んだら、適当なところでgitの`feature/[適当なブランチ名]`ブランチにコミットして、githubにpushしておいてください。

numata-1740, 2024-05-08:
**ｚｘ１２０のＩＭＵ配置**
大変遅くなり申し訳ありません

IMUセンサとGNSSの取り付け位置を調査した画像により
zx120 について unity ローカルプロジェクトにてGNSSとIMUを追加しました

https://github.com/pwri-opera/OperaSim-PhysX/tree/feature/gnss_imu_numata

以下が主な変更点です．

body 上に gnss を１基追加
GeoCooodinateSysytem もプロジェクトに追加（座標は土研のDXフィールド）
bucket, arm, boom, body にそれぞれIMU を追加
IMUの正確な配置（特に向き）は分からないので、見た目で適当に配置しています．

Github のローカルリポジトリ作成に手間取りました。
今回の修正についてコミットをせず、
ローカルで作成したプロジェクトのUnity データを一括してローカルリポジトリに
アップロードしコミットしています。
まだまだ、トレーニング中ですので何か問題や不具合があれば
ご連絡ください。

numata-1740, 2024-05-08:
> @numata-1740 IMUの型番がわかったら、「データの出力レート」についても可能であれば合わせてください。
下記の型式となります
出力レートは　：125kbit - 1Mbit/sec となっています。
今回のコミットには反映していませんので、次回のコミットに反映させる予定です。
[image]
[image]

yosuke, 2024-05-08:
@numata-1740 
ブランチは作成されているものの、コミットがありませんでした。
gitのブランチ作成、コミット、githubへのpushは、重要なオペレーションなので練習してできるようになっておいてください。
「IMUのレート」ですが、ビットレートではなく「サンプリングレート」が知りたい＆合わせて欲しいです。

numata-1740, 2024-05-09:
@yosuke 様

大変遅くなり申し訳ありませんでした

下記に修正したソースファイルをアップロードしました。

https://github.com/pwri-opera/OperaSim-PhysX/tree/gnss_imu_numata

尚、IMUのレートについては反映していません

何か問題や不具合があればご連絡ください。

numata-1740, 2024-05-20:
@yosuke 様
zx120 ショベルについて
feature/gnss_imu_numata　ブランチにアップロードしました
修正内容：
IMUセンサ：body、boom、arm　を追加しました
また、gnss を追加しました
尚、IMU、gnssのサンプリングレートについては反映していません。
何か問題や不具合があればご連絡ください。

numata-1740, 2024-05-21:
@yosuke 様
zx120 ショベルについて
トピック名とトピック毎のデータレートを
筑波大学共同研究で取得した実機の bag データファイルから
再生して取得したデータレートをご連絡いたします。

表１．GNSS IMU センサ 通信トピックと通信サンプリングレート調査
No. | 名称 | 部位 | トピック | サンプリングレート[msec] | 備考
-- | -- | -- | -- | -- | --
1 | GNSS |   | gnss_odom | 100 |  
2 | IMU | rotator | g2_imu | 10 |  
3 | IMU | boom | g2_imu | 10 |  
4 | IMU | arm | g2_imu | 10 |  
5 | IMU | bucket | g2_imu | 10 |  

[image]
[image]
[image]
[image]
[image]

よろしくお願いいたします。

numata-1740, 2024-05-22:
@yosuke 様
zx120 ショベルについて
feature/gnss_imu_numata　ブランチにアップロードしました
修正内容：
トピック名を実機の通信ログに合わせて修正しました
また、IMU、gnssのサンプリングレートについて同様に修正しました。
何か問題や不具合があればご連絡ください。

maopotcha, 2024-09-17:
本件、バケット用のIMUは課題が残留しておりますがこれは以下で扱うことにしてクローズにします
https://github.com/pwri-opera/OperaSim-PhysX/issues/61

closed 2024-09-17 by maopotcha