pwri-opera/OperaSim-PhysX #48: バケット掘削時の粒子挙動が不適当

maopotcha, 2024-05-02:
### 概要
バケット掘削時の粒子の挙動が現実との乖離大
- 問題1：掘削動作をさせてもほとんどの粒子がバケット内に入らない
- 問題2：バケットから溢れた粒子が、バケットから一定距離離れるまで（バケットの外側であっても）粒子として振舞ってしまう
- 問題3：溢れた粒子が多く集中している箇所については、その後バケットが離れた時に隆起してしまう
- 問題4：粒子が飛び散る距離が大きくなる（跳躍する）時がある

[image]

### 目的
粒子（土砂）の挙動を現実とシミュレータでできる限り合致させる

### やること
- 上記問題1～4の原因究明
- 原因を取り除く対策の実施

### その他
- 各問題の原因考察
    - バケットが地面から受ける反力の計算結果が現実よりも大きい
    - 粒子を維持する範囲が大きすぎる（バケットの外側についてはもう少し小さくする）
    - 粒子が生成される瞬間に、粒子同士の反力が過大に計算されてしまう

yosuke, 2024-05-08:
Physics Debugを有効にすると現象が解析しやすい。
粒子径がある程度大きい（このスクリーンショットの場合は0.1m）場合はある程度の量の粒子がバケット内に残ってくれている。
[image]

yosuke, 2024-05-08:
boomの関節が制御負けしてしまっている傾向あり。各xDriveパラメータを10倍するとそれなりに追従してくれる。

yosuke, 2024-05-10:
粒子が跳ねて飛んでいくように見える現象については、粒子の新規出現位置が他の粒子と被ってしまっており、瞬間的に大きな反発力が発生するため。
本来は空きスペースを探して粒子を出現させるべきだが、粒子が多くなると空きスペースを探す計算コストが大きくなってしまうため、これまでは実装を避けていた。
ただ、最近導入したBurst compile機能を使えば高速に計算できる可能性があるので、試しに実装してみたい。

yosuke, 2024-05-10:
粒子は跳ねて飛んでいくように見える現象、簡易にできる対処として粒子の最大速度を設定してみたところ、かなり綺麗に解決できてしまった。
計算負荷もほとんど増加しないので、この対処で良いかもしれない。

yosuke, 2024-05-14:
跳ねて飛んでいく現象を抑えるために今まで粒子が重めに設定されていたが、妥当な値に調整。粒子間力にも依存するが、以下のような出来型になるように調整完了。

[image]

TaroABEJido, 2024-07-04:
/feature/realistic_soil_particle_behavior の粒子挙動について，添付掘削動作のように粒子が地面に戻る際に，バケットに対して反力が発生しているようです．修正してみます．
[image]

TaroABEJido, 2024-07-04:
BucketRocks.cs 140 行目 の bucket と粒子の距離の閾値を 0.5 -> 1.0 にすると気持ち反発しなくなったような気がします．
ただ掘削後の地面に粒子が溜まり，山のようになる現象は若干起きているような感じもします
[image]

TaroABEJido, 2024-07-05:
* 粒子の挙動調整のための Issue が新たに必要かも？

closed 2024-07-05 by TaroABEJido