pwri-opera/OperaSim-PhysX #50: ツールチップを追加する

yosuke, 2024-05-10:
各スクリプトのプロパティ部分に説明のツールチップを追加する。

ツールチップの追加方法については以下を参照：
https://ekulabo.com/tool-tip-attribute

AGX版にはすでに追加されており、こちらも参照してください：
https://github.com/pwri-opera/OperaSim-AGX/blob/main/Assets/Machines/DumpTruck/Scripts/DumpSoil.cs

numata-1740, 2024-05-23:
@yosuke 様
修正内容の確認ですが
Web画面で使用される、マウスオーバーのポップアップ表示を
Unityツール内で表示されるように修正を行うと理解しています。
具体的には、例えば、zx120モデルでは、base_link、body_link、boom_link、arm_link、bucket_link　について
Inspector タブで、各Component の変数についてマウスオーバー表示することでよろしいでしょうか。
作業手順ですが、まずは、
１）１個、マウスオーバーのポップアップ表示をテスト修正する
２）実行テストする
３）修正項目を調査して、一覧表にする：変数名と範囲値、表示コメント
４）確認頂いてから修正を実施する
と考えております。
よろしくお願いいたします。

numata-1740, 2024-05-24:
@yosuke 
サンプルテストで、ツールチップを作成
ローカルプロジェクトで作成しテストしてみました。
以下になります
サンプルソース
[image]
テスト結果
[image]

numata-1740, 2024-05-28:
@yosuke 様
ツールチップ表示項目について調査の中間報告です。
調査方法に誤りがありましたら教えてくさい。
よろしくお願いいたします。
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/NUMATA~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/NUMATA~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

No. | クラス名 | public定義変数 | 参照先 | ツールチップ内容
-- | -- | -- | -- | --
1 | VesselController | DumpTopicName | c30r,     ic120 | これはDumpのトピック名です。
2 | dump_joint |  
3 | BucketRock | manager | zx120,     zx200 |  
4 | terrain | これは掘削地形です。
5 | JointPosController | setpointTopicName | arm_link     body_link,     boom_link,     bucket_limk,     zx120,     zx200 | これはジョイントのトピック名です。
6 | initTargetPos | これは初期値の回転角度　rad　です。
7 | DiffDriveController | robotName | c30r,     ic120,     zx120,     zx200 | シミュレーションロボットモデル名
8 | TwistTopicName | トピック名
9 | OdomTopicName | GNSSトピック名
10 | childFrameName | チャイルドフレーム名
11 | pGain | ｐゲイン
12 | iGain | ｉゲイン
13 | dGain | ｄゲイン
14 | torqueLimit | トルク上限
15 | brakeTorque | ブレーキトルク
16 | maxLinearVelocity | 最大リニア
17 | maxAngularVelocity | 最大アンギュラー
18 | publishMessageInterval | パブリッシャーメッセージインターバル
19 | JointStatePublishe | topicName | ic120,     zx120,     zx200 |  
20 | enableJointEffortSenso |  
21 | publishMessageInterval |  
22 | OdomPublisher | robotName | base_link,     ic120 |  
23 | topicName |  
24 | childFrameName |  
25 | publishMessageInterval |  
26 | SoilParticleSettings | enable | zx120,     zx200 |  
27 | particleVisualRadius |  
28 | partileStickDistance |  
29 | stickForce |  
30 | m_AngleOfRepose |  
31 | m_ReposeJitter |  
32 | syncPeriod |  

</body>

</html>

yosuke, 2024-05-29:
@numata-1740 
表だと全てを網羅しているかがわからないので、assets/scriptsフォルダ内のファイルを直接編集して適当なgit branchを作成してpushしてください。

numata-1740, 2024-06-04:
@yosuke 様
ツールチップ表示項目について修正が終了しました。
ブランチの：feature/tool_tip_numata をアップロードいたしました。
変数について理解不足のため仮修正となります。
修正内容についてチェックお願いいたします。
よろしくお願いいたします。

maopotcha, 2024-09-04:
一通りの自作スクリプトのユーザ設定パラメータに追加完了したためクローズします

closed 2024-09-04 by maopotcha