pwri-opera/OperaSim-PhysX #53: Windowsでビルドできない

AKota3, 2024-06-05:
概要
Build SettingsでPlatformをAndroidにしてビルドしようとしてもビルドができない。Windowsでもビルドできない。

目的
AndroidやWindowsでビルドしてapkファイルやexeファイルで出力する。

やること
ビルド時に出るエラーの対処。

原因の考察
ビルドすると
Assets\Scripts\HeightmapFromTexture.cs(5,3): error CS0246: The type or namespace name 'MenuItemAttribute' could not be found (are you missing a using directive or an assembly reference?)
と
Assets\Scripts\HeightmapFromTexture.cs(5,3): error CS0246: The type or namespace name 'MenuItem' could not be found (are you missing a using directive or an assembly reference?)
というようなエラーメッセージが出力される。
https://teratail.com/questions/330085
を参考にしてエラーの出たスクリプトを新たに作成したEditorというファイルに移動させたところ、このエラーは出なくなった。しかし、次に
Assets\Scripts\SoilParticleSettings.cs(50,14): error CS0103: The name 'AssetDatabase' does not exist in the current context
というエラーが出た。そこでSoilParticleSettings.csもEditorフォルダーに移動させた。この操作の後、ビルドするとAssets\Scripts\BucketRocks.csに関して
Assets\Scripts\BucketRocks.cs(126,13): error CS0103: The name 'SoilParticleSettings' does not exist in the current context
というようなエラーが出たため、同様にもEditorフォルダーに移動させた。するとWindowsでのビルドは可能となったが、同サイトやunityのhttps://docs.unity3d.com/ScriptReference/UnityEditor.html
によるとこの方法ではEditorの中身は参照されなくなるようなので根本的な解決方法となっているかは分からない。

yosuke, 2024-06-12:
@AKota3 @maopotcha 
`HeightmapFromTexture`はビルド後は使わない機能なので、Editorフォルダに移動するのは適切な対応と思います。
`SoilParticleSettings`と`BucketRocks`は移動してしまうとビルド後に土が掘れなくなってしまうので、`AssetDatabase`クラスを使わない別の方法で対応した方が良いと思います。
正しいやり方を調べてみます。

yosuke, 2024-06-12:
compute shaderは、`Resources.Load`を使ってロードするのが本来のやり方だが、Thermal.computeが別パッケージに入っているので、それをロードするためにAssetDatabaseを利用する必要がある。Thermal.computeのカスタマイズが必要な局面もまもなく出てきそうなので、Resourcesフォルダにコピーを置いてしまうのが良いか。
https://qiita.com/kodai100/items/f75ef809e7a1c56dd27f

yosuke, 2024-06-14:
@AKota3 (@maopotcha)
`AssetDatabase`を使わない形にコードを修正してみました。masterブランチにpushしてあります。
`HeightmapFromTexture`のみEditorフォルダに動かす形でpull requestを送り直してもらえるでしょうか？
うまくビルドできなかったら教えてください。

AKota3, 2024-06-14:
HeightmapFromTextureのみEditorフォルダに動かしたものをpull requestをお送りいたしました。

yosuke, 2024-06-14:
@AKota3 ありがとうございます。マージしました。

maopotcha, 2024-06-17:
本件、解決したようなのでクローズします
@yosuke  @AKota3 ご対応ありがとうございました

closed 2024-06-17 by maopotcha