pwri-opera/OperaSim-PhysX #62: IMUのセンサシミュレーション結果に爆発的ノイズが生じる

maopotcha, 2024-07-26:
### 概要
IMUのacceleration出力に7～10秒に一回程度、爆発的なノイズが生じる。
[image]

### 目的
爆発的ノイズがセンサシミュレーション出力にのらないようにする

### やること
・原因究明
・原因の除去

### その他

numata-1740, 2024-08-06:
発生現象について原因調査を実施しました。
調査内容について不備や問題がありましたらご指摘ください。
【調査報告１】
１）発生現象
　　IMUのacceleration出力に7～10秒に一回程度、爆発的なノイズが生じる。
　　図１：
[image]

２）現象把握（再現）
　　ブランチ：OperaSim-PhysX-feature-gnss_imu_numata
　　ROS-unity 環境で/zx120/boom/g2_imu をモニタ監視
　　動画１：

３）原因調査１
　　ノイズの発生しているIMUのacceleration　xyz について
　　unity でデバッガーを起動してノイズ発生時の値でブレークポイントを
　　設定して、データの調査解析を実施
　　図２：ノイズ発生時のacceleration　xyz

　　図３：unity imu_sensor モジュールの値でブレークポイントを設定
[image]

　　図４：ブレークポイントでの図形調査

　　図５：年月日時分秒secsを基準にROS-unity、imuSencorsのデータを分析
　　ノイズ発生までのunity imuSensors内の変数データをデバッグログで取得し
　　ノイズ発生後の現象について調査解析する

　　
　　図６：データ整理
　　unity 起動からノイズ発生までのimuSensors内の変数データを一覧表に整理

　　図７：解析図
　　unity 起動からノイズ発生までのimuSensors内の変数データをグラフにして解析
　　

４）原因の把握１
　　データ整理から、ノイズ発生時の先頭トリガーは、transform position x と把握
　　図８：

　　図９：

解析結果から、imuSensors内の入力データであるtransform positionの元データを
調査解析が必要

numata-1740, 2024-08-07:
【調査報告２】
調査報告１を踏まえ、ノイズ現象の発生元である入力データを調査

５）原因調査２
　　ノイズ発生元 transform position x のデータ取得
　　図１０：

６）原因の把握２
　　ノイズ発生元 transform position x のデータ解析
　　図：

７）原因調査３
　　図：

８）原因の把握３
　　図：

maopotcha, 2024-08-23:
zx120よりもic120の方が爆発的ノイズの発生頻度が高い
クローラを模擬させるためのwheel colliderを無効化すると、ノイズが消えることを確認
（ただし、この場合はクローラがTerrainに沈み込む）
[image]

maopotcha, 2024-08-23:
IMU Sensor内の処理にローパスフィルタを追加したところ、ノイズレベルが低減することを確認
[image]
これはこれであった方がいいように考える
ただし、ノイズが消えるわけではなく対処療法なので原因究明を継続する

maopotcha, 2024-08-26:
zx120およびic120オブジェクトに用いているFixed Joint Component中のMass Scaleを減少する又はConnected Mass Scaleを増大すると振動がおさまる現象を確認。現時点でデメリット・背反を把握しきれていないため、これを対策とするかどうかについては別途検討して判断する。
[image]

closed 2024-09-17 by maopotcha