pwri-opera/OperaSim-PhysX #89: ZX200のマニピュレーション操作がうまくいかない

kasahara-san, 2025-03-04:
**現象**
シミュレータ上のZX200について、front_cmdが発行されているのにも関わらず機体が動かない

**理想**
front_cmdの指令値がシミュレータ上の機体の動作に反映されるようにする

**再現方法**
コミットID 90bd4cfについて、widows PC上で起動し、Ubuntu PCとROS-TCP-Endpointを介して接続後、Ubuntu PCでzx200_ros2を起動する。そしてRViz上でモーションプラニングにおけるゴール姿勢を指定し、実行する。すると、front_cmd等の指令値が発行されるものの、シミュレータ上の機体はこれに基づいて動作しない。コミットID 9ee0a81では意図したとおりの動作をすることを確認済。

kasahara-san, 2025-03-23:
ご対応ありがとうございます。

デフォルトがvelocity入力となっているとのことでしたので、以下のコマンドでzx200操作用のノード群を立ち上げて再度試してみたのですが、機体は動きませんでした...

ros2 launch zx200_unity zx200_standby.launch.py command_interface_name:=velocity

なので色々試してみました。まず、前のバージョン(コミットID 9ee0a81)においてzx200_ros2からvelocity入力指定で操作する際には以下のトピックが吐かれていることを確認しました。( この際はUnity上の機体は正常に動作しました )
・ /zx200/swing/cmd
・ /zx200/arm/cmd
・ /zx200/boom/cmd
・ /zx200/bucket/cmd
ただ、これらはposition入力用のトピックと理解しています。velocity入力用のトピックは/zx200/tracks/cmd_velかと思うのですが、こちらは発行されていませんでした。

従って、シミュレータ操作時にはzx200_ros2がposition入力のみにしか対応できていないものと思われます。

なお、/zx200/front_cmdは発行されていませんでした。何らかの理由で別PCからのトピックが見えていたものと思われます。誤った報告をしてしまいすみません。

上記を加味して、OperaSim-PhysXのmainブランチの最新バージョンで、OperaSim-PhysX側のControl Type Annotation スクリプトの"Control Type"をposition指定に変更し、zx200_ros2をposition指定で実行してみました。すると正常に動作することを確認しました。従って、今回の原因はzx200_ros2がposition入力のみにしか対応していなかったことによるものだと理解しました。

なお、補足ですがOperaSim-PhysX側にてControl Type Annotation スクリプトの"Control Type"をposition指定にしておけば、 zx200_ros2をposition・velocity・effortのどれで起動しても/zx200/swing/cmd・/zx200/arm/cmd・/zx200/boom/cmd・/zx200/bucket/cmdは発行されるようなのでUnity上の建機は動作しました。

yosuke, 2025-03-25:
zx200_unityパッケージのzx200_standby.launch.pyは、command_interface_nameの設定に寄らずposition制御になってしまうので、それも原因かもしれません。
https://github.com/pwri-opera/zx200_ros2/blob/main/zx200_unity/launch/zx200_standby.launch.py#L50-L51

```
ros2 launch zx200_bringup vehicle.launch.py command_interface_name:=velocity
```
でやってみてもらえますでしょうか？

kasahara-san, 2025-03-28:
実行してみたところ、以下のログが出力され、うまく起動できませんでした。
[move_group-2] You can start planning now!
[move_group-2]
[INFO] [spawner-5]: process has finished cleanly [pid 6920]
[INFO] [spawner-4]: process has finished cleanly [pid 6918]
zx200_bringupは実機操作用のパッケージで、velocity制御及びeffort制御に設定可能ですがOperaSim-PhysXには対応してないようです

yosuke, 2025-04-11:
こちらで動かしたところ、zx200_bringupの方で正常動作することは確認できました。

launch fileは以下のものを使っています。

https://github.com/pwri-opera/opera-ros2-rocker/blob/main/all.launch.xml

必要パッケージは以下を参照してください。

https://github.com/pwri-opera/opera-ros2-rocker/blob/main/Dockerfile

launch直後に肘が伸びすぎてしまって逆運動学の計算がエラーで落ちてしまう状態だったので、「Joints」のパネルで正常な範囲に調整してから動かしています。

[image]

`/zx200/front_cmd`のpub/subは以下のようになっていました。

```console
$ ros2 topic info -v /zx200/front_cmd
Type: com3_msgs/msg/JointCmd

Publisher count: 1

Node name: uac_velocity_hw
Node namespace: /
Topic type: com3_msgs/msg/JointCmd
Endpoint type: PUBLISHER
GID: 01.0f.aa.1f.77.1d.c8.1b.00.00.00.00.00.00.22.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: zx200front_cmd_RosSubscriber
Node namespace: /
Topic type: com3_msgs/msg/JointCmd
Endpoint type: SUBSCRIPTION
GID: 01.0f.aa.1f.6d.1d.63.3f.00.00.00.00.00.00.2e.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```

yosuke, 2025-04-18:
アームが伸び切った状態でmoveit2が起動してしまう件は、以下のプルリクエストがマージされたら解決。
https://github.com/pwri-opera/zx200_ros2/pull/8

kasahara-san, 2025-04-19:
ご返答遅くなってしまい申し訳ありません
ご連絡頂いたとおり試したところ、正常に動作することを確認いたしました。
ご対応ありがとうございました。

closed 2025-07-28 by kasahara-san